wled/WLED #3400: Sun Times Return Zero

tonyn0, 2023-09-29:
### What happened?

Sometimes, sun times are calculated as zero (depending on location, I hear). Requesting that "use real math" is implemented as standard.

### To Reproduce Bug

Be in wrong place at wrong time. ;)

### Expected Behavior

Calculations are always valid.

### Install Method

Binary from WLED.me

### What version of WLED?

b6

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-30:
Partial workaround is by custom compiling using `-D WLED_USE_REAL_MATH`.
The drawback is that image size will increase and also a few bytes of precious RAM will be missing.

The former may be an ESP32 issue and the latter ESP8266 issue.

softhack007, 2023-09-30:
I'll propose a proper fix tomorrow. It seems that reverting `_t` math to standard functions in getSunriseUTC() will make it work. Reason is that _t functions are slightly faster, but only use deliver low accuracy values.

From Discord
> Wondering if I found a bug. This was working fine on v5. My sundown keeps reverting to 0's. I can get the sundown to pop up once I change something, but after I save it and leave the page, it reverts back.

[image]

softhack007, 2023-10-13:
fixed in 0.14.0

closed 2023-10-13 by softhack007

softhack007, 2023-12-23:
See https://github.com/Aircoookie/WLED/issues/3601.
A more robust solution will be availeable with release 0.14.1