wled/WLED #3403: Adding a new segment sometimes disables the Color palette controls

RadianM, 2023-09-30:
### What happened?

Selecting "+ Add segment" in the UI sometimes locks-in "Color palette not used" irrespective of the subsequent selection of an effect where the effect flags indicating that a palette is used.

### To Reproduce Bug

Take a fresh install of WLED, set it up and restart. The default **Solid** effect will be set in **Segment 0**. Select **+ Add segment** and confirm. Now select this new **Segment 1** in preparation for selecting an effect mode and notice that whatever effect is selected, the palette area remains unpopulated with the notice **Color palette not used**.

### Expected Behavior

I would expect the selection of any effect for the new segment to populate the palette area if the effect has the "uses palette" flag set.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b6 (and earlier)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Simply deleting the "bad" segment and adding it once again usually clears the problem but once it's happened when adding a new segment, subsequently added segments are also guaranteed to result in disabled palettes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-30:
Can you clarify (take screenshot before adding and after adding a segment) which segment is "main"?
It should be lightly colored.

RadianM, 2023-09-30:
Step one: Flash WLED to an ESP32 @ https://install.wled.me and load up the UI:

[image]

Step two: Select **+Add segment** and click the tick icon to create it:

[image]

Step three: Select newly created Segment 1 (light grey highlight) and notice minimally populated palette area:

[image]

I assume this is what you mean by the "main" segment? i.e. the segment that runs the selected effect?

At this point, no matter what effect is selected for the new segment, no color controls are available.

blazoncek, 2023-09-30:
That is normal and expected behaviour since the selected effect is Solid. It does not have palette or secondary color associated with it.
What is more, you are creating overlapping segments which is something of advanced configuration and not recommended for beginners.
And on top of that the segment capabilities determine if color selectors or palettes, heck, even effects, can be displayed.
None of this is well documented unfortunately.

I do not consider this a bug and my own testing show correct behaviour (as intended).

RadianM, 2023-09-30:
With great respect, have you actually tried it? The solid effect on show there is  red herring.  It should still render the colour wheel anyway. Irrespective of the effect selected when a new segment is added, a different effect can be selected for the overlay (yes, I'm overlaying effects which I don't consider to be advanced stuff!)

Just fire up WLED and try making an overlay from bouncing balls, drips, glitter etc. and see what happens. Then consider if it's expected behavior for new overlay segment to occasionally disable the color controls. Thanks.

blazoncek, 2023-09-30:
I was not able to reproduce the behaviour until I used an ESP with only one segment and created overlapping "default-sized" segment (spanning entire strip) (I have about 25 most of them with more than one output/default segment).
In such case capabilities are not correctly calculated and segment, unless its bounds change, reports only On/Off capabilities (i.e. no color).

As a workaround change start or stop address and you can then revert it back.

Unfortunately I was not yet able to pinpoint the source of the cause.

RadianM, 2023-09-30:
Your helpful workaround of changing the start or stop LED does indeed seem to cure it - so that has to be a clue!

To explore thus a little further I just did the following: added a segment one LED shorter than the full width then add another at the full width then add a fourth also at full width and that last one failed.

Segment 0=30
Segment 1=29 (OK)
Segment 2=30 (OK)
Segment 3=30 (fail)

So the miscalculation does just seem to be when the UI tries to come up with a start/stop range when the underlying segment is full width.

blazoncek, 2023-09-30:
Thank you but I am well aware what the cause is that's how I proposed the workaround.
The issue is *why* the code that configures capabilities does not work for new segment when there is already a similar segment present (with full strip width).
But don't worry, I'll figure it out. Meanwhile don't make overlapping segments of full width.

RadianM, 2023-09-30:
That's good to hear. I happen to think effect overlays are one of the more exiting aspects of WLED but they seem to be under-appreciated. The potential to combine effects such as flames, smoke and fireworks - each in their own layer (as I have managed to hack together by tweaking some existing effect code) is producing such marvels as a custom bonfire night party.

closed 2023-10-01 by blazoncek