wled/WLED #3404: Add legend or hover tooltip for filters

tablatronix, 2023-09-30:
These are a bit ambiguous, it would help new users to know what these meean. A simple alt tag might do the trick, or a legend somewhere in UI

[image]

tablatronix, 2023-09-30:
I am not even sure what these flags mean. Not sure its even documented fyi.

https://kno.wled.ge/features/effects/

blazoncek, 2023-09-30:
If you figure out correct CSS I will gladly add it.

tablatronix, 2023-09-30:
Looks like there is a tooltip class

```html
<span id="sliderLabel0" class="tooltiptext">Effect speed</span>
```

tablatronix, 2023-09-30:
This will require a lot of adjustments because of the sticky and overflow visibility and z index etc.

[image]

blazoncek, 2023-09-30:
Fill the search field and test on iOs/Android as well, please.

blazoncek, 2023-09-30:
When you have it ready you can make a PR.
https://github.com/Aircoookie/WLED/wiki/How-to-properly-submit-a-PR

If you do not want to bother just submit what changed.
Thank you.

tablatronix, 2023-10-13:
Doing it now

tablatronix, 2023-10-13:
Need to figure out how to compile first, having an issue with 
(project configuration file): 'No option 'ldscript_1m0m' in section: 'common''

tablatronix, 2023-10-13:
How tf do I upload the index.htm, I have been uploading spiffs and nothing changes..

blazoncek, 2023-10-13:
https://kno.wled.ge/advanced/custom-features/#changing-web-ui

closed 2024-01-04 by blazoncek