wled/WLED #3405: Segment 0 brightness fader can become inaccessible

RadianM, 2023-09-30:
### What happened?

After setting up two segments, then subsequently deleting segment 1 caused the UI to hide the Segment 0 slider which had been set to less than full brightness. This setting then becomes inaccessible.

### To Reproduce Bug

Add a new Segment (length doesn't matter) then dim down Segment 0 with its brightness slider.
Delete the new Segment you just added (or select **Reset segments**)
Notice that segment 0 brightness remains low but the Slider is now hidden.

### Expected Behavior

If the UI makes Segment 0 brightness slider inaccessible, it ought to be restored to full at that time. Leaving the effect running at reduced brightness with no immediate way of increasing it is very unexpected. Especially so, given the naming of the **Reset segments** button.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b6 (and earlier)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-01:
Consequence of #2984

RadianM, 2023-10-01:
> Consequence of #2984

So I think there's a case to be made for setting Segment 0 brightness to max when the slider is made inaccessible. At the very least it could be actioned when selecting the **Reset segments** button. Without such a fix, the two simple steps I outlined above can make WLED appear inoperative.

blazoncek, 2023-10-01:
I have the fix ready, until it is out set segment brightness to max before deleting segments.

RadianM, 2023-10-01:
OK great. Looking forward to a release that addresses this and #3403  Resolving both these issues will make building multilayer patterns less prone to malfunctions.

closed 2023-10-01 by blazoncek