wled/WLED #3410: Reset segments between presets in a playlist

benjamw, 2023-10-02:
**Is your feature request related to a problem? Please describe.**
I have some presets that have segment data contained in the preset (e.g.- a preset that has two alternating solid colors, with 5 pixels each, which can only be done with segments), and other presets that do not contain segment data (for ease of sharing, and/or they were imported from a shared library). When these presets are placed in a playlist with a random order, if the segmented preset plays, and then the non-segmented preset plays, it will only change the first preset, leaving alternating bands of color in the lights (from the segmented preset).

**Describe the solution you'd like**
It would be nice if there were an option when creating a playlist (especially a random one), where it would reset the segments to a default single segment (or allow a preset as an input that sets the wanted segments for users that have complicated segment settings) when switching between presets in the playlist. That way, it clears out any preset segment settings, and allows the subsequent presets to play as intended without the history of other presets hanging around.

**Describe alternatives you've considered**
Storing segment settings in all presets is an option, but having segments stored in presets makes them less sharable and less modular.

blazoncek, 2023-10-02:
Not technically possible.
You will need to save presets *with* segment data included. You can always load and save with "Use current state" and "Include segment bounds" to update effects on a device to include segments data. Keep a device (or a copy of presets) with segmentless presets for modifications. My recommendation would be a copy of presets in a prettified format for ease of modifying.

benjamw, 2023-11-26:
Ok, I'll add segment data to all my presets. Thanks.

closed 2023-11-26 by benjamw