wled/WLED #3411: Stutter/ Lag every 10s when adding Homeassistant Integration

Hannes-Beckmann, 2023-10-02:
### What happened?

As soon the as the esp is discovered and configured in Homeassistant exactly every 10 seconds the loop takes ~110 ms to finish once. This is visible in the animations as stutter. In the serial monitor there is a warning seen below.
I have updated Homeassistant today:

Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest

I am using the audioreactive build. With the normal build the issue seems to be not as bad with a loop time of ~35 ms.

### To Reproduce Bug

Flash WLED with debug enabled and observe the serial monitor. Every 30s the debug info is printed and the loop maximum loop duration is a few milliseconds. Then configure the discovered integration in Homeassistant and observe the serial monitor. Now every 10 s there is the warning mentioned below.

### Expected Behavior

Homeassistant causes no stutter and the animation remains smooth

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b6 env: esp32dev_audioreactive

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
19:40:07:458 -> Not-Found HTTP call:
19:40:07:460 -> URI: /presets.json
19:40:07:462 -> WS FileRead: /presets.json
19:40:07:571 -> Loop took 116ms.
19:40:07:572 -> Usermods took 37ms.
19:40:07:574 -> Strip took 1ms.
19:40:07:606 -> JSON buffer locked. (17)
19:40:07:613 -> JSON buffer size: 3220 for request: 3
19:40:07:616 -> JSON content length: 1921
19:40:07:624 -> JSON buffer released. (17)

19:40:17:477 -> Not-Found HTTP call:
19:40:17:479 -> URI: /presets.json
19:40:17:480 -> WS FileRead: /presets.json
19:40:17:587 -> Loop took 114ms.
19:40:17:588 -> Usermods took 54ms.
19:40:17:590 -> Strip took 0ms.
19:40:17:621 -> JSON buffer locked. (17)
19:40:17:628 -> JSON buffer size: 3220 for request: 3
19:40:17:632 -> JSON content length: 1921
19:40:17:638 -> JSON buffer released. (17)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-02:
Known issue caused by HA polling WLED too often depleting resources on ESP.
Nothing can be done on WLED side. Reduce polling from HA or remove integration and use MQTT.

numanair, 2023-10-09:
@Hannes-Beckmann Are you using the nmap tracker in HA?

Hannes-Beckmann, 2023-10-09:
> @Hannes-Beckmann Are you using the nmap tracker in HA?

No, I do not. I am using the FritzBox integration (https://www.home-assistant.io/integrations/fritz/) to track devices.

closed 2024-05-12 by github-actions[bot]