wled/WLED #3415: EST Server Time Incorrect For Daylight Savings Time

greensleeves149, 2023-10-04:
### What happened?

In Australia, the Australian Eastern Time (AEST) Zone changes for Daylight Savings on the first Sunday of October and again on the first Sunday in April. 

On the first Sunday of October each year, AEST clocks move forward one hour. "

On the first Sunday of April each year, AEST clocks move back one hour. 

As I write this on the 4th October and the first Sunday of October has just passed, the time currently is 14:38, but WLED server time shows 13:38. 

This means that all my lights on fixed timers come on one hour later as the server time is incorrect. 

### To Reproduce Bug

Just by checking the current AEST server time on WLED against the current AEST should reproduce the problem. 

### Expected Behavior

On the FIRST Sunday of October each year, the server time for AEST needs to advance forward one hour. 

On the FIRST Sunday of April each year, the server time for AEST needs to decrease back by one hour. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-10-05:
seems to be a duplicate of 
* https://github.com/Aircoookie/WLED/issues/2820

greensleeves149, 2023-10-06:
Thank you for the information softhack007. You're spot on. I updated to 0.14.0-b6 and checked the time and it seems perfect. Apologies for that, but thank you for your direction.

greensleeves149, 2023-10-06:
Yes- spot on. This was my fault. I didn't update the software so the problem continued. Sorry about that.

closed 2023-10-06 by blazoncek