wled/WLED #3429: Error adding to Home Assistant

tuley24, 2023-10-05:
### What happened?

Using the WLED integration in home assistant I am getting an error when I try to manually add the host.

### To Reproduce Bug

Using an ESP32-S3 on v0.14.0-b6, attempt to add to a home assistant instance using supported WLED integration. I also tried v0.14.0-b5 and am running into the same issue.

### Expected Behavior

Integration adds without error.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b6_ESP32-S3_8MB, build 2309241

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
Logger: aiohttp.server
Source: /usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py:403
First occurred: 12:54:59 PM (3 occurrences)
Last logged: 3:09:04 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 63, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 177, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 293, in async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 394, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/wled/config_flow.py", line 40, in async_step_user
    device = await self._async_get_device(user_input[CONF_HOST])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/wled/config_flow.py", line 117, in _async_get_device
    return await wled.update()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 275, in update
    self._device = Device(data)
                   ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 694, in __init__
    self.update_from_dict(data)
  File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 751, in update_from_dict
    self.info = Info.from_dict(_info)
                ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 391, in from_dict
    (filesystem := Filesystem.from_dict(data)) is not None
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 330, in from_dict
    percentage=round((used / total) * 100),
                      ~~~~~^~~~~~~
ZeroDivisionError: division by zero
```

### Anything else?

Reading through the log it looks like something might be wrong with the disk space reading. On the web interface, my filesystem size in the info tab shows "0/0 kb (NaN%)" in case that is what is causing my error.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-10-05:
```
File "/usr/local/lib/python3.11/site-packages/wled/models.py",
 line 330, in from_dict 
 percentage=round((used / total) * 100), 
                                   ~~~~~^~~~~~~ ZeroDivisionError: division by zero
```

This looks like a division by zero somewhere in HA. Maybe report the problem to HA developers?

softhack007, 2023-10-05:
> On the web interface, my filesystem size in the info tab shows "0/0 kb (NaN%)" in case that is what is causing my error.

Oops, overlooked this one. 0/0kb should not happen, as it means you don't have a littlefs partition in flash memory. 
Could it be that your -S3 only has 4MB flash?

tuley24, 2023-10-06:
> > On the web interface, my filesystem size in the info tab shows "0/0 kb (NaN%)" in case that is what is causing my error.
> 
> Oops, overlooked this one. 0/0kb should not happen, as it means you don't have a littlefs partition in flash memory. 
> Could it be that your -S3 only has 4MB flash?
> 
> Maybe try to use this online installer, which also has "nightly builds" for -S3 with 4MB: https://wled-install.github.io/

Thanks I'll give that a try and report back.

And I'm using a seedstudio XIAO ESP32S3 that shows to have both 8M of PSRAM and 8MB of Flash so I figured it would be okay but maybe that's cutting it close.

tuley24, 2023-10-06:
I installed the 4MB version from the link you provided and I still am seeing the same issue. I flashed back to the 8mb version and I noticed I'm getting "error 19 filesystem error" which is likely related.

Also I initially installed WLED through a UF2 file I created with esptool as I couldn't get any other method to work out of the box. After I got it up and running I reinstalled the official bin since it was now working.

softhack007, 2023-10-06:
> Also I initially installed WLED through a UF2 file

Stil seeing "0/0 kb (NaN%)" ?
If yes, then most likely you still have a bad filesystem, if your reinstalled the official bin via OTA upload. UF2 creates a FAT32 partition, while WLED expects LittleFS.

As it seems more like an installation problem and not a bug in WLED, I'd suggest to close this bug report. 

You might want to join our [WLED discord](https://discord.gg/KuqP7NE) to get help and support from the WLED community.

tuley24, 2023-10-06:
Thanks for the info, I'll close this issue and try another install method.

closed 2023-10-06 by tuley24