wled/WLED #3430: Adjustable delay time for power on - problem with relay and a flash at power on

DieNummerEins, 2023-10-06:
**Is your feature request related to a problem? Please describe.**

My setup:
- 1 ESP32 WLED controller (always on with its own PSU)
- 10 meters SK6812 rgbww led strip
- 6 PSUs every 2-2,5 meters as power injection on 230V line
- relay (SSR) => inrush current limiter in front of 230V line (as protection for fuse not to blow out, what it definitely does by more than 3 PSUs)
- relay is connected to WLED to power off/on the PSUs

My problem:
If I turn lights on (e.g. "turn on" button in GUI) I see first a short flash with random colors. It seems as the PSU needs a bit of a second to provide stable 5V but WLED already forces the LED to light up.

**Describe the solution you'd like**
Like in issue 812 (see below) a "delay on power on event" would help. If I click on "power on" (via GUI) or "power on" command is send, WLED first actuate the relay and only after a time to be set (delay time) forces the LED to lighten up.

**Describe alternatives you've considered**

I tried a workaround via a playlist (first a black preset for 0,2s duration, then the desired light configuration). It works fine, but this has disadvantages like described in
https://github.com/Aircoookie/WLED/issues/346
And, I think, I cannot turn light on with this playlist vie GUI (turn on is just turn on without playlist/preset)

In a thread I cannot find right now there were a solution considered to make "turn on/off" event more generally. I mean turn off via GUI and the same turn off on boot. On boot I can start a playlist/preset, but not with GUI-turn-on-button. This feature would make a small difference at the playlist/preset-workaround for my entire problem. I plan to have 30-40 WLED inside/outside my house, not to turn on/off via GUI isn't nice. With the feature to customize the turn on/off event (not just at boot), I can hide my problem from using experience, it is still a workaround (based on the missing delay option) but it is more user friendly. I will submit this as a separate feature request.

**Additional context**
I commented a similar thread:
https://github.com/Aircoookie/WLED/issues/812

...and explained the situation in discord issues channel on October 5th/6th 2023

blazoncek, 2023-10-12:
My recommendation is to use Multi relay usermod and connect relay switch to a LED GPIO output.
This will achieve desired result without changing code at the expense of additional relay (or some other switch).
Multi relay usermod has adjustable on delay.

doronazl, 2023-12-12:
> My recommendation is to use Multi relay usermod and connect relay switch to a LED GPIO output. This will achieve desired result without changing code at the expense of additional relay (or some other switch). Multi relay usermod has adjustable on delay.

how does it work exactly ? 
currently on the led menu i put in relay gpio - say 15 in my case
then when i turn the wled button on and off, it does the same with the relay 

but with multirelay usermod, im confused
if im choosing under led menu - the relay as gpio 15, then when i go to the usermod multirelay menu page and try to choose relay0 as gpio 15 in order to control the time delay, gpio 15 shows up as used, so my question is , the relay0 function on the multirelay and the relay function on the led preferences menu, do they work the same ?

like, if i dont choose a relay on the led prefernces menu at all, and choose the gpio 15 on relay0 under usermod multirelay, would that work the same? turn on and off along with the wled on/off button ?
if not then what is the solution here ?

softhack007, 2024-11-02:
* https://github.com/Aircoookie/WLED/pull/4060

Allows to add a delay at startup before powering on LEDs and other units. Only available in custom builds by `-D WLED_BOOTUPDELAY=500` (or any other delay in milliseconds)

closed 2025-01-17 by DedeHai