wled/WLED #3431: Make "power on event" more generally and customizable

DieNummerEins, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
At feature request #3430 I described my problem and a possible solution. In this context it would help, to make "power on event" more generally and customizable. For power on at boot I can choose presets/playlists, for power on via GUI-button I can not.

**Describe the solution you'd like**
It would be nice, if you can customize the power on event (GUI-button) and add presets/playlists to it.

**Additional context**
Feature request #3430

blazoncek, 2023-10-17:
> It would be nice, if you can customize the power on event (GUI-button) and add presets/playlists to it.

How would you tackle *on* and how *off* event? That would require 2 presets or just one correctly formatted.
Would you want to change API as well?

DieNummerEins, 2023-10-17:
In GUI (App or Browser) there is a power button. If it's off and I push it then "on event" is fired, vice versa.

If I can customize the on and the off event separately (with presets/playlists and maybe cascade of both) that would be awesome. 

It would also be nice to fire both events (on and off) via HomeAssistant for example. Now I have a switch, therefore it would be same as GUI-power-button (if it's off and I push then on event and vice versa).

But I can imagine situations where it is necessary to just fire specificly the off event, for example to make sure light is off (if I am not home or at every midnight to safe energy). Maybe it is possible already, I'm new to this stuff.

What to do in API I cannot say, I don't know the details.

DieNummerEins, 2023-10-18:
What is GUI override?

In "LED Preferences" it is possible to decide, wether the LEDs should be turned on or not and which preset to apply.
[image]

It would be great to have the preset/playlist option for GUI power button. It is the smaller idea of the above (turn on event more generally and customizable).

blazoncek, 2023-10-18:
I know it would be great but is difficult to implement with current code.

DieNummerEins, 2023-10-18:
So is the "power on delay" (for special problems with relay usage) apparently ;-). Well, maybe the current code will be rewritten someday and those changes will be more easily.

blazoncek, 2023-10-18:
That's totally unrelated (and different problem for which I already gave you a workaround). 😄

DieNummerEins, 2023-10-18:
Yes, you are right, it is a different challange, but they have the same in common: "difficult to implement with current code".

Unfortunately the workaround with 2 relays is too complicated. My solution is the following: My wall switches (4 button type) are connected to a ESP32 with ESPHome. I use a separate ESPHome-controller for my wall switches because first they are located somewhere in the middle of the room ,the wled-strip starts elsewhere and so the wled-controller is too far away. And second sometimes there is a power outlet underneath the wall switches, so the ESPHome-controller has an additional relay for it. Above all there is HomeAssistant, the ESPHome-controller uses presets/playlists to turn on wled-strips.

Additionally I will rework my structure with this page:
https://kno.wled.ge/basics/compatible-controllers/
LevelShifter and Resistance/EMC-Filter is maybe something important for stabilization.

DieNummerEins, 2023-10-18:
Anyway, all of you guys make a great job and I am very thankful for WLED and it's continually improvement.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek