wled/WLED #3438: BlinkNlights support

mallory98e, 2023-10-12:
Would this be possible to add blinknlights support where patterns can be uploaded into the micro, with possible animations, changing patterns, realtime text from pc etc?

softhack007, 2023-10-12:
link? I never heard about this tool.

asciireign, 2023-10-16:
I guess he means that.
[Project Blinkenlights Talk 2023](https://media.ccc.de/v/camp2023-57001-project_blinkenlights)
Just dropping this here since it would overlap with what wled enthusiasts probably find cool. :sunglasses: 

Basically more 2d matrix effects and other fancy stuff.
They even color calibrated the leds. :+1:

mallory98e, 2023-10-17:
Yes exactly: https://www.youtube.com/watch?v=BP29JE6jgAg

I have saw something like this in a hackerspace built into a model building, it was completely custom made.

Moving some patterns would be nice but also computer control where you can send text, date, maybe even upload gif on the fly etc, I realize this would take a lot of space on ESPs already but maybe you could make it modular.

blazoncek, 2023-10-17:
For realtime PC control etc, you can use LedFX, xLights, Jinx etc.