wled/WLED #3440: TypeError: gId(...) is null

ThomasCr, 2023-10-13:
### What happened?

After update to WLED 0.14.0 I get the following Error message:

[image]

### To Reproduce Bug

I can't say, why I get this error. It get displayed on the main page.

### Expected Behavior

No error message or a better way to know whats wrong.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Source-Map-Fehler: Error: request failed with status 404
Ressourcen-Adresse: http://192.168.99.201/rangetouch.js
Source-Map-Adresse: rangetouch.js.map

TypeError: gId(...) is null
    populateSegments http://192.168.99.201/:221
    readState http://192.168.99.201/:221
    requestJson http://192.168.99.201/:221
    promise callback*requestJson http://192.168.99.201/:221
    onLoad http://192.168.99.201/:221
    loadPalettesData http://192.168.99.201/:221
    onLoad http://192.168.99.201/:221
    setTimeout handler*onLoad/</</< http://192.168.99.201/:221
    loadFX http://192.168.99.201/:221
    loadFX http://192.168.99.201/:221
    onLoad http://192.168.99.201/:221
    loadFXData http://192.168.99.201/:221
    loadFXData http://192.168.99.201/:221
    onLoad http://192.168.99.201/:221
    loadPalettes http://192.168.99.201/:221
    loadPalettes http://192.168.99.201/:221
    onLoad http://192.168.99.201/:221
    t http://192.168.99.201/:7
    setInterval handler* http://192.168.99.201/:7
    EventListener.handleEvent* http://192.168.99.201/:7
    EventListener.handleEvent* http://192.168.99.201/:7
192.168.99.201:221:6827
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ThomasCr, 2023-10-14:
Hi again. Today in the morning I got the idea to reboot the device and voila, the error is gone. I just hope it stays away.

blazoncek, 2023-10-14:
You've got 404 on `rangetouch.js`, that is not expected and has caused the error you were seeing.
It should fix itself on page refresh.

ThomasCr, 2023-10-15:
No, I get the 404 only on the `rangetouch.js.map` - I was checking before on Network debug tab of the browsers developer tools.

ThomasCr, 2023-10-15:
And the `gId(...) is null` is back again. I think I have to reset the config and set the thing again up from scratch... but its not like it should go....

blazoncek, 2023-10-15:
Please show network trace from your browser's developer tools.

ThomasCr, 2023-10-16:
for now I dont have the error again. I will report back when it happen again.

ThomasCr, 2023-12-12:
Hi again - the error is still present - also when I was quite for a while (I dont use it so mush - its in my boys room and they reported that is is not working).

[image]

[image]

ThomasCr, 2023-12-12:
ok, I removed the "Enable custom CSS" in the options - the 404 about the skin.css is gone. But the `gId(...) is null` is the same.

ThomasCr, 2023-12-21:
today I updated to v0.14.1-b1 - hope its magically gone...

ManuelW77, 2024-01-21:
I've got the same error after upgrading to 0.14.1

Did you find a solution?

delhite2, 2024-02-24:
was having the exact same issue,
step by step troubleshooting helped me resolve it.
In my case it was a bad preset, it was basically a preset with no effect which was loaded when PIR sensor detected no movement.
So double check all the presets and if necessary re-do them.

closed 2024-07-01 by github-actions[bot]

dmak, 2024-12-10:
Got the same issue on v0.14.0:
```
TypeError: gId(...) is null
    populateSegments http://10.0.0.118/:221
    readState http://10.0.0.118/:221
    requestJson http://10.0.0.118/:221
    promise callback*requestJson http://10.0.0.118/:221
    onLoad http://10.0.0.118/:221
    loadPalettesData http://10.0.0.118/:221
    onLoad http://10.0.0.118/:221
    setTimeout handler*onLoad/</</< http://10.0.0.118/:221
    loadFX http://10.0.0.118/:221
    loadFX http://10.0.0.118/:221
    onLoad http://10.0.0.118/:221
    loadFXData http://10.0.0.118/:221
    loadFXData http://10.0.0.118/:221
    onLoad http://10.0.0.118/:221
    loadPalettes http://10.0.0.118/:221
    loadPalettes http://10.0.0.118/:221
    onLoad http://10.0.0.118/:221
    t http://10.0.0.118/:7
    setInterval handler* http://10.0.0.118/:7
    EventListener.handleEvent* http://10.0.0.118/:7
    EventListener.handleEvent* http://10.0.0.118/:7
```
Probably is reproducible only on slow WiFi connections.

DedeHai, 2024-12-10:
is this still an issue in 14.4 or even latest 0.15 betas?

sLill, 2024-12-13:
> is this still an issue in 14.4 or even latest 0.15 betas?

Yes. I have the same error in 0.15. And for me, my presets never load- so I presume it's the same root cause.

blazoncek, 2024-12-13:
Please show browser's console log and network trace.

sLill, 2024-12-13:
Was able to resolve this by re-flashing the same 0.15.0 image, but making sure to atleast visit the locally hosted colors webpage before disconnecting it from my pc. Previously, I had just enabled the ethernet port and immediately disconnected. Not sure why that fixed it. Perhaps something to do with browser caching?

dieseltravis, 2024-12-16:
I am running 0.15.0 on a WT32-S1, I get this same error in Firefox:
[image]
```
downloadable font: Glyph bbox was incorrect (glyph ids 9 19) (font-family: "WIcons" style:normal weight:400 stretch:100 src index:0) source: data:font/woff2;charset=utf-8;base64,d09GMgABAAAAAAnUAAsAAAAAE1AAAAmFAAGZmgAAAAAAAAAAAAAAAAAAAAAAAAAABmAAgXwRCAqcYJZIATYCJANwCzo … o1+fFCBb6hvuLz8lMROwfjPN8iar90RCCiRCJr3ugqHf6LqgUYYs5hzvu9tMIOUr/xpvRsNVvdZ/p+mB8n7V2Spo0T+aRhPpNhsNFOqxoE2u0suqTipgx58IJA0AAAA=
TypeError: gId(...) is null
    populateSegments http://192.168.1.83/:1
    readState http://192.168.1.83/:1
    requestJson http://192.168.1.83/:1
    promise callback*requestJson http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    setTimeout handler*loadPalettesData/< http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    loadPalettesData http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    loadFX http://192.168.1.83/:1
    loadFX http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    setTimeout handler*onLoad/</< http://192.168.1.83/:1
    loadFXData http://192.168.1.83/:1
    loadFXData http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    loadPalettes http://192.168.1.83/:1
    loadPalettes http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    onload http://192.168.1.83/:1
192.168.1.83:1:78146
TypeError: gId(...) is null
    populateSegments http://192.168.1.83/:1
    readState http://192.168.1.83/:1
    requestJson http://192.168.1.83/:1
    promise callback*requestJson http://192.168.1.83/:1
    setTimeout handler*requestJson/< http://192.168.1.83/:1
    promise callback*requestJson http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    setTimeout handler*loadPalettesData/< http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    getPalettesData http://192.168.1.83/:1
    setTimeout handler*getPalettesData/< http://192.168.1.83/:1
    promise callback*getPalettesData http://192.168.1.83/:1
    loadPalettesData http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    loadFX http://192.168.1.83/:1
    loadFX http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    setTimeout handler*onLoad/</< http://192.168.1.83/:1
    loadFXData http://192.168.1.83/:1
    loadFXData http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    loadPalettes http://192.168.1.83/:1
    loadPalettes http://192.168.1.83/:1
    onLoad http://192.168.1.83/:1
    onload http://192.168.1.83/:1
192.168.1.83:1:78146
```

In Chrome I get different errors:
[image]
```
2(index):1 TypeError: Cannot read properties of null (reading 'classList')
    at populateSegments ((index):1:92305)
    at readState ((index):1:103337)
    at (index):1:108176
```

sLill, 2024-12-16:
> Please show browser's console log and network trace.

I wasn't able to screenshot the logs because I resolved the issue, but they were exactly the same as the ones @dieseltravis posted above.