wled/WLED #3442: Allow ESP-NOW Remote to trigger different actions: cycle presets, change modes

ooglek, 2023-10-14:
I am SUPER excited about the ESP-NOW-based Remote option. I fly a paramotor, and when I want to change the settings, I have to get out my phone, connect to Wifi... it's a bit tedious.

And one always wants lots of options!!! Reading that ESP-NOW Remote only supported 4 presets, I was still excited, but wondered if there could be more.

**SOLUTION DESIRED**
Customize the button configuration to allow for virtually unlimited means of controlling WLED.

The remote currently allows you to select only Presets 1-4, set Brightness/Dim, turn on/off.

I'd love to see some power options for Presets 1-4, as well as the Up/Down button:

1. MODE POWER: When first powered on, the WLED device will be in "Preset Mode" for the remote. Pressing Up/Down will go through the presets, starting with the default preset, and move up/down through the Preset modes, looping back to the end/beginning depending on which way you press, allowing you to access your favorites quickly, though with maybe a few more button presses than just a single button press.

In this scenario, Buttons 1-4 now function as Mode Selectors:
 - 1: Preset Mode
 - 2: Effects Mode
 - 3: Color Mode
 - 4: Random Mode
 
 In Preset Mode, you can cycle through the Presets. If default is preset 7, hitting up (brighter) goes to preset 8. If 9 presets, on Preset 9, Up goes to preset 1. On Preset 1, down goes to preset 9. 
 
 In Effects Mode, you can go through all the Effects, starting with the currently selected one, in alphabetical order.
 
 In Color Mode, you can go through all of the Color options, starting with the currently selected one, in alphabetical order.
 
 In Random Mode, each Up/Down press will select a different Effect and Color at random. And may also randomly select the speed.
 
2. SPEEDSTER: Same as MODE POWER, except Random Mode is replaced with Speed Mode -- Increase or Decrease the speed of the currently running Effect.

3. ULTIMATE KONAMI: Using the 6 buttons (1-4 up down), you can program all sorts of insanity.
  - 1-1-up-down-down-up: 1-1 is Preset, up-down-down-up is 1001 in binary, representing decimal 9, so Preset 9
  - 1-2-4-up-down-down-down: 1-2 is Effects, up-down-down-down is 8 in binary, plus 16 times 3 (4 minus 1), select the 56th Effect
  - 1-3-3-down-down-up-up: 1-3 is Color, down down up up is 3 in binary,  plus 16 times 2 (3 minus 1), select the 35th color
  I am sure there is a better way, just off the top of my head.

**ALTERNATIVES**

I could keep using my mobile device, but it is a bit of a pain.

I looked for ESP-NOW remotes with more buttons, but there does not yet seem to be a market there.

blazoncek, 2023-10-14:
That's already available in 0.15 alpha.
Binaries not yet available.

ooglek, 2023-10-16:
Woot! Sorry, did not realize. I'll go look for the issue related to the feature, or look for the code. Thanks!

ooglek, 2023-10-16:
Am I looking at the wrong code?

I just looked at `main` as well as the `0_15` branch and it looks like `UP` and `DOWN` are still hardcoded to brightness, without an option to change its functionality.

https://github.com/Aircoookie/WLED/blob/f329a5950e6a24eb13864d6ad83b078457150d25/wled00/remote.cpp#L63-L73

The `OnDataRecv()` function in `wled00/remote.cpp` seems hardcoded to the presets and brightness, which is exactly the opposite of what my Feature Enhancement is suggesting. 

I am suggesting that Button 1-4 and Up/Down allow for some more interesting options to more comprehensively control WLED.

1. On first boot, "Preset" mode is active. Let's say Preset 7 is the default on Power up. Pressing `UP` would move WLED to Preset 8. `UP` again would move to preset 9. `UP` again would move to Preset 0 (or 1). 
2. Pressing the `TWO` button, You now are able to cycle (`UP`/`DOWN`) through the Effects.
3. Pressing the `THREE` button, You are now able to cycle (`UP`/`DOWN`) through the Colors.
4. Pressing the `FOUR` button, using `UP` and `DOWN` you can have an Effect and Color selected at Random.
5. Pressing the `ONE` button takes you back to `Preset Mode.`

blazoncek, 2023-10-16:
You are missing the point.
Implementation in 0_15 allows for any action to be set to a button.

ooglek, 2023-10-16:
I'm missing the code I guess. Which branch should I be looking at to see how this is implemented?

ghost-rider4, 2023-12-15:
Hi, is it possible to control the wled software through espnow via another esp32 rather than the wiz-remote. If yes what would be the program for the same?

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.
Use `remote.json` in a similar format as `ir.json`.

closed 2024-03-22 by blazoncek