wled/WLED #3443: ESP8266 crashes with WLED 0.14 when extensive using wbe interface

wled-install, 2023-10-15:
### What happened?

- ESP8266
- WLED 0.14.0 installed from install.wled.me

When web interface is extensively used (fast change between different effects by clicking them randomly, faste change of set color, fast reload web interface, also after opening "info" page), the ESP8266 rebotts at some time. It might happen after few seconds or after many minutes. Happens in WLED-AP mode as well as when WLED is integrated in local WiFi network. The same is tested with 0.13.3 and it is no issues with this older version.

### To Reproduce Bug

- Install WLED 0.14.0 on ESP8266
- Click fast randomly to different effects and/or reload web interface pages fast/switch between different pages of web interafce

### Expected Behavior

No reboot

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Example 1:
Terminal log file
Date: 14.10.2023 - 23:43:07
-----------------------------------------------
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
Exception (3):
epc1=0x4010180d epc2=0x00000000 epc3=0x00000000 excvaddr=0x400054d1 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd60 end: 3fffffd0 offset: 0150
3ffffeb0:  000038a8 00000715 3ffe8994 401018c6  
3ffffec0:  00000000 40254a1e 3fff722c 3fff2e10  
3ffffed0:  3fff5e3c 00000020 3fff79cc 40101c14  
3ffffee0:  3fff781c 3fff2da8 3fff5e6c 40253c84  
3ffffef0:  00000000 3fff2da8 3fff780c 40253f02  
3fffff00:  00000000 00000000 3fffff20 3fff781c  
3fffff10:  00000000 00000432 3fff2da8 4024c06a  
3fffff20:  3fff77fe 3fff77fe 00000432 3fff28a0  
3fffff30:  3fff03a0 3fff2bb0 3fff03a0 0174c935  
3fffff40:  3fff73cc 00014269 08b43958 0174c972  
3fffff50:  3ffffec0 00000000 ec8b4395 3fff37c0  
3fffff60:  3fff2fc0 00000001 3fff2fbc 402390ea  
3fffff70:  3fff0270 00000000 3fff3794 40239154  
3fffff80:  3fff0270 00000000 3fff3794 40247662  
3fffff90:  00000000 001c006e 4026b17c 3fff37c0  
3fffffa0:  3fffdad0 00000000 3fff3794 3fff37c0  
3fffffb0:  3fffdad0 00000000 3fff3794 40265354  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100dfd  
<<<stack<<<
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000d3c80
~ld
    g  s  n| $ l`c  |r l o  g <0>l`  { d d  Ada

-----------------------------------------------
Date: 14.10.2023 - 23:45:19
End log file

Example 2:
Terminal log file
Date: 15.10.2023 - 08:28:14
-----------------------------------------------
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
Exception (9):
epc1=0x40254458 epc2=0x00000000 epc3=0x00000000 excvaddr=0x040a841d depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffd0 offset: 0150
3ffffee0:  00000000 3fff2da8 3fff6a3c 40254491  
3ffffef0:  3fff6a3c 3fff2590 3fffff20 4022fe08  
3fffff00:  00000441 3fff28a0 3fffff20 4022fe08  
3fffff10:  00000000 00000441 3fff2da8 4024c001  
3fffff20:  402a7a1f 00000441 3fff331c 3fff28a0  
3fffff30:  3fff03a0 3fff2bb0 3fff03a0 4025c710  
3fffff40:  00004e48 000670ca 7b645a1c 07726a1e  
3fffff50:  3ffffec0 6814a8c0 03d70a3d 3fff37c0  
3fffff60:  3fff2fc0 00000001 3fff2fbc 40239102  
3fffff70:  3fff0270 00000000 3fff3794 4023916c  
3fffff80:  3fff0270 00000000 3fff3794 4024767a  
3fffff90:  00000000 001c006e 4026b198 3fff37c0  
3fffffa0:  3fffdad0 00000000 3fff3794 3fff37c0  
3fffffb0:  3fffdad0 00000000 3fff3794 4026536c  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100dfd  
<<<stack<<<
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000d3c90
~ld
    g  s  '| d l c  |; l o  o <0>l`  { d l  Ada

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000d3c90
~ld
    n {  o| l d`c  |s d g  ' <0>d`  r l d  Ada

-----------------------------------------------
Date: 15.10.2023 - 08:37:21
End log file

Example 3:
Terminal log file
Date: 15.10.2023 - 08:51:33
-----------------------------------------------

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000d3c90
~ld
ŒãìƒgäÛsƒû'|ìdŒl c›Ÿ|;“lœoàƒoã<0>l „ã{“dÄlŒþAda
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-15:
That is "normal" behaviour as ESP8266 has very limited resources and AsyncWebServer (a component/library WLED uses) tries to fulfill every web request made. This leads to RAM depletion and consequently a restart.

Please use ESP32 if you intend to "bombard" WLED with plenty of requests per second.

wled-install, 2023-10-15:
OK, if the issue is known and there is no other implications, i will close it.

closed 2023-10-15 by wled-install

blazoncek, 2023-10-15:
Unfortunately AsyncWebServer has no locking mechanism and dynamically allocates memory for every request made.
In case of rapid sequence of requests it allocates all available memory leaving no free RAM for other stuff and WLED crashes before being able to fulfill requests made.