wled/WLED #3446: ESP32-POE ETH01 no connection with APA102 configurated

Baalthorun, 2023-10-16:
### What happened?

The Ethernet connection fails when the LED Configration is set to APA102.

### To Reproduce Bug

Set the LED configuration to APA102

### Expected Behavior

working Ethernet connection with activated APA102 configuration. Other LED strips such as SK6812 work perfectly

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

Other

### Anything else?

I have seen issue #2542
https://github.com/Aircoookie/WLED/issues/2542
According to this, the error should be fixed. With Release 14.0 it does not seem to work.

Sometimes it works until reboot but then the interface is dead.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-16:
My own Ethernet set-up works correctly.
Please provide your configuration (cfg.json) so we can diagnose better.

Baalthorun, 2023-10-16:
[wled_cfg.json](https://github.com/Aircoookie/WLED/files/12913797/wled_cfg.json)

Board: https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

blazoncek, 2023-10-16:
What happens if you de-configure WiFi?
If possible please get a debug build from @srg74 and post complete boot sequence.

Baalthorun, 2023-10-16:
Disabling WLAN does not change anything. Except that the AP is activated. I left it on to make it easier to test different GPIO configurations.

Sorry my programming experience and Github experience is not very extensive. How do I get the dbug build? I have not found a way to contact @srg74 directly.

blazoncek, 2023-10-16:
Apparently he does not have Ethernet debug builds.

blazoncek, 2023-10-16:
Here it is: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Latest/Debug_builds

Baalthorun, 2023-10-16:
hope I have done everything right

[putty.log](https://github.com/Aircoookie/WLED/files/12922144/putty.log)

blazoncek, 2023-10-17:
Everything seems to be in order. And the code has not been changed in quite some time.
Unfortunately I have no ideas left.

closed 2024-04-23 by github-actions[bot]