wled/WLED #3448: Connect failures.

bert269, 2023-10-16:
### What happened?

After flashing 14.0 I cannot connect to device after setting wifi credentials.
Both on android app and windows/browser.

### To Reproduce Bug

RE-flashed 14.B6 and it works fine. Flash 14.0 and cannot connect. flash 14.b6 and it works fine again

### Expected Behavior

I was hopeing that 14.0 was propperly tested before it was released.

### Install Method

Binary from WLED.me

### What version of WLED?

NOw back on 14.b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-10-16:
Hi, thanks for your report.

> I was hopeing that 14.0 was propperly tested before it was released.

Actually it was. One month before release we stopped adding any new features, and beta users were testing everything.
So there must be something special with your setup.

* How much flash does your 8266 have? 1MB, 2MB, or 4MB? 
* AP mode, or full WiFi connection?
* It would also help to have your cfg.json and presets.json (from beta6 that seems to work for you), for reproducing your issue.

bert269, 2023-10-16:
4M device(s).
I am on full wifi.
Where can I get the cfg.json and preset.json files?

Sorry for the noobness - I am new to ESP8266

bert269, 2023-10-17:
I just repeated the same step on a device that was 14b6.  - re-flashed it witl 14.0. The moment after it is flashed, it does nlot load the web-ui - just shows "Loading UI"
I have to physically remove it from the WLED app, and re-assign it.
I also receive scipt errors (red pop-up) quickly after loading.
If I do not remove/re-add it to the app, it will shows: "Loading UI" and not getting anywhere - that is what I was referring to.

blazoncek, 2023-10-17:
That may be an app issue. Do not use app when reporting issues, use regular browser.

bert269, 2023-10-17:
I understand not to "use the app to report issues" - but after the upgrade to 14.0 the app is not working anymore

blazoncek, 2023-10-17:
The app is separate product with its own repository.
You may want to try out WLED Native app, available in Apple App Store and Google Play Store.

bert269, 2023-10-17:
I will use the WLED Native instead - thanks

nsaneadmin, 2023-10-18:
> I just repeated the same step on a device that was 14b6. - re-flashed it witl 14.0. The moment after it is flashed, it does nlot load the web-ui - just shows "Loading UI" I have to physically remove [...]

What phone/ Android Version are you running?

bert269, 2023-10-18:
Android 14.

closed 2024-04-23 by github-actions[bot]