wled/WLED #3449: Add only Hour and Minute to Scrolling Text Segment Label

knarfd, 2023-10-17:
Scrolling Text Segment recognition is nice. But would it be ok, that only add a Hour and Minute recognise, so that you can place the hour and minute to anywhere on the 2D Segment where you want (e.g. have 2 Segments, one with Hour and one with Minute). I suggest the labels #HH and #MM.

My Feature Request is, that i want to build a clock with an 16x16 matrix, but with the smallest font size it starts to scroll when the hour is >=10. (with 0-9 hour it just fits).

Adding a #HH and #MM Label for the segment will be very helpful (currently #HHMM and alternatives allready exist), so i can create two scrolling segments. One for Hour, one for Minute

The following code at FX.cpp should be ok, to add to the other else if statements after line 5948:
`    else if (!strncmp_P(text,PSTR("#HH"),3))   sprintf_P(text, zero?PSTR("%02d")          :PSTR("%d"),         AmPmHour);
    else if (!strncmp_P(text,PSTR("#MM"),3))   sprintf_P(text, zero?PSTR("%02d")          :PSTR("%d"),         minute(localTime)); `

Alternative would be to have some "non scrolling" Text. (but i think as scrolling automatically calculates if it needs to scroll it would be easier to just add those else if statements).

Thanks a lot.

blazoncek, 2023-10-17:
You are welcome to make a [PR](https://github.com/Aircoookie/WLED/wiki/How-to-properly-submit-a-PR).

knarfd, 2023-10-17:
ok.. thanks.. i tried.. hopefully i did the right thing ( #3450 )

blazoncek, 2023-10-17:
You also closed the PR.

knarfd, 2023-10-17:
ah sry.. reopned it. i'm not so well known to github.. thanks anyways

closed 2024-01-04 by blazoncek