wled/WLED #3454: WiFi refuses to connect after reboot

chilicheech, 2023-10-17:
### What happened?

When I install wled 0.14.0 on an ESP-WROOM-32 (ESP32 DEVKITV1) (https://www.amazon.com/dp/B08D5ZD528?ref=ppx_yo2ov_dt_b_product_details&th=1) everything works fine. It connects to WiFi, controls the lights, everything is cool.

When I reboot/power cycle the ESP it refuses to connect to WiFi. The AP comes up and I'm able to connect to it via the AP. I can scan and find WiFi networks and update the settings. But it will no longer connect to WiFi.

### To Reproduce Bug

- install wled 0.14.0 on an ESP-WROOM-32 (ESP32 DEVKITV1) (https://www.amazon.com/dp/B08D5ZD528?ref=ppx_yo2ov_dt_b_product_details&th=1)
- connect it to a wifi network
- reboot/power cycle the ESP
- observe that it no longer connects to the wifi

### Expected Behavior

It should connect to wifi after a reboot/power cycle.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-18:
It works on 32 of my devices. I would suspect something is odd in your network set up.

chilicheech, 2023-10-18:
This looks to be an issue with maybe the wifi antenna. I moved the esp board to a slightly different location and it worked fine. It's my first time using these boards and it seems the wifi reception is a bit sensitive. Gotta make sure to position the board at a location with excellent wifi reception.

closed 2023-10-18 by chilicheech