wled/WLED #3456: Google Pixel Phones won't load Web UI

nsaneadmin, 2023-10-18:
### What happened?

Since I update to version 14 the WLED Native app and chrome android browser gets stuck on Loading Web UI. I've reflashed back to 13 and everything will work again on my phone, but once I flash back to 14 the web UI gets stuck again. I have no issues loading it on other phones and PC in my home just my Pixel Phones. It could also be an issue with Android 14 the other devices I have used are on android 13 and lower.

If I open a browser with IP.com/settings every works fine from my phone also it just won't load the main UI. 

### To Reproduce Bug

Use and Google Pixel 8 or 6 with Android 14

### Expected Behavior

I expect the web UI to load, but it just sticks at loading. 

### Install Method

Binary from WLED.me

### What version of WLED?

Installed version: WLED 0.14.0 (ESP32-D0WDQ5 build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Please let me know if you need any screenshots or any other information. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-18:
Please connect your phone to your PC, open browser's developer tools and monitor what's happening in network tab and console log.
Without his information we cannot determine where the problem lies.

nsaneadmin, 2023-10-18:
[192.168.90.5.zip](https://github.com/Aircoookie/WLED/files/12985481/192.168.90.5.zip)
 attached is a har file

nsaneadmin, 2023-10-18:
> Please connect your phone to your PC, open browser's developer tools and monitor what's happening in network tab and console log. Without his information we cannot determine where the problem lies.

[image]

nsaneadmin, 2023-10-18:
> [192.168.90.5.zip](https://github.com/Aircoookie/WLED/files/12985481/192.168.90.5.zip) attached is a har file

[image]

blazoncek, 2023-10-18:
Thank you. It seems "onLoad()" does not get executed when it should. Possibly due to delay.
Please tap on the settings cog and then back. Or try refreshing several times.

nsaneadmin, 2023-10-18:
> Thank you. It seems "onLoad()" does not get executed when it should. Possibly due to delay. Please tap on the settings cog and then back. Or try refreshing several times.

After hitting refreshing for a few minutes it finally loaded! 
[image]

blazoncek, 2023-10-18:
If you can compile yourself, please try compiling `0_15` branch where I tried to implement request throttling to prevent ESP8266 being drained of resources. It may help in this particular case.

nsaneadmin, 2023-10-18:
Okay I will try and give that a shot. Will need to do some research on compiling

browsean, 2023-10-18:
I'm having the same issue on a Samsung S23 Ultra with Android version 13.

nsaneadmin, 2023-10-20:
> If you can compile yourself, please try compiling `0_15` branch where I tried to implement request throttling to prevent ESP8266 being drained of resources. It may help in this particular case.

I've compiled the esp32dev for version 0.15.0-a0 and still seem to run into the same issue. 

If I use any other PC or Android phone I don't seem to have the issue. :(

pakkman12, 2023-12-14:
FYI not sure if it will help but Pixel Fold working fine also on Android 14

blazoncek, 2023-12-14:
Fire up developer tools for Android browser and check for possible error messages.

closed 2024-06-05 by github-actions[bot]