wled/WLED #3458: Constant board reboots

SergioLuxx, 2023-10-18:
### What happened?

Using the websocket API, I send LED control commands to the esp8266 every 0.1 - 0.5 milliseconds. On firmware 0.14.0, the lamp freezes every 5 seconds and reboots, which is not observed on firmware 0.13.0 and the same congestion of the communication channel.

### To Reproduce Bug

Send many websocket requests in a very short period of time

### Expected Behavior

I expected stable operation of the lamp on firmware 0.14 under heavy load on the API. This stability is available on firmware 0.13

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Please fix it

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-18:
Please upgrade to ESP32 if you intend to hammer ESP with requests every few milliseconds.
Resource demands have almost doubled in 0.14 and ESP8266 is no longer able to keep up.
There is nothing that can be done except stripping features.

scottmclesly, 2023-10-28:
Check if your controllers reset pin is the same as default data pin of WLed.
I ended up installing 0.13 audio - it worked just fine. 

A Default data pin selector on WiFi selection screen would be awesome to have.

closed 2024-04-23 by github-actions[bot]