wled/WLED #3467: Active preset after 0.14.0

R1DEN, 2023-10-21:
### What happened?

Before the update if I turned off the light (by using the button in the upper left corner, or via API), the previously selected preset was still active/selected/reported in Home Assistant.
After the update, as soon as I turn off the lights - the preset is disabled/inactive/not reported in Home Assistant.
The settings themselves are not changed, but the preset is deselected in the UI.

### To Reproduce Bug

1) pick any preset
2) turn wled off via the button

### Expected Behavior

The preset should still be active (both in UI and in Home Assistant) as it was previously.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-21:
Not a bug but intentional behaviour.
Since presets can contain "on" state any change in UI (or from API) that causes state change (including "on" state) is considered *preset invalidator* (active preset is no longer valid).

R1DEN, 2023-10-21:
@blazoncek so was it considered a bug in all the previous versions?

R1DEN, 2023-10-23:
It seems to me that reverting it to the way it was provides more flexibility to the API user, as you could always check if "on" is false or not to determine if the preset is actually working or not.
But without the preset being reported in the "off" state - you cannot predict what would be the active preset in case you turn WLED on. Furthermore when turning it of and on again the preset is not restored, despite the light output being the exact same as it was when the preset was active.

UrTekNerd, 2023-10-27:
I discovered a new issue with WLED 0.14.0. First I would like to say Thank you for WLED. I love it. The issue I discovered is when every you run a playlist it wipes out all your segments. If I stop the playlist and just set a color with an effect it's fine. I had three segments and as soon as I started the playlist all my segments were gone. I was hoping to run 4 different playlists on 4 different segments.  Thank you.

blazoncek, 2023-10-28:
> I had three segments and as soon as I started the playlist all my segments were gone. I was hoping to run 4 different playlists on 4 different segments.  Thank you.

Segments are stored in presets, not playlist. Playlist does not remove segments, presets can.

If you are using ESP8266 segments can be removed by core functionality due to low memory condition. In such case ESP may also diconnect all clients and reconnect to the WiFi.
Please check how much available heap there is using Info page.

closed 2023-12-31 by blazoncek