wled/WLED #3468: fade from RGB preset  to CCT preset issue in fading

marcobrianza, 2023-10-21:
### What happened?

I am using a RGB+CCT configuration.
The transition from a RGB preset to a CCT preset starts with the wrong CCT.

### To Reproduce Bug

make a preset "RED" with a RGB color RED, set the CCT intensity to 0 leave the CCT slider in the middle. (you just see RED)
make a preset "COLD" with all RGB to 0 and a CCT with intensity to MAX and CCT slider to COLD.

if you transition from "RED" to "COLD" the warm LED turn on during the transition.

a workaround is to set the slider to COLD in the RED preset.
this would be an issue if I need to transitio to a WARM preset. and I would need to make a RED preset with the slider to WARM.

### Expected Behavior

In transitions to CCT the intensity value should be considered to apply the start value

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

using this controller
https://www.athom.tech/blank-1/wled-rgbcct-analog-and-digital-light-strip-controller

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-21:
As you already discovered in your workaround setting desired CCT in your starting preset will correctly transition colors.

Unfortunately, if you have warm white color enabled in your starting preset it *will* transition to cold. To put it the other way, the warm white will illuminate at the start of transition.
It is impossible to do it otherwise (or I am unable to envision how this could be done).

marcobrianza, 2023-10-22:
Hi, 
The software is perfectly able to perform preset fading on 3 channels on RGB. I see no technical reason why it should not work with 5.
Presets are a set of value for the different channel that are used as destination for a transition. Of course the destination does not have to be dependant on the source values of the transition. The workaround I fond is just to investigate the cause of the bug, that is the source values are not correctly calculated because it is not considering the intensity parameter (being 0 both cold and warm should start from 0, and if a channel also end to 0 it should never light up).
I also see that the software is able to consider the 2 CCT values to apply CCT in a static situation.  I think that the same approach can be used to set the values for transition start.
If you can please point me to the code area where this happens I can have a look if it is a simple thing or it needs an architectural change.

blazoncek, 2023-10-22:
> The software is perfectly able to perform preset fading on 3 channels on RGB. I see no technical reason why it should not work with 5.

You are welcome to implement that if you think it is *that simple*.

marcobrianza, 2023-10-25:
Never said is a simple task, just "technically possible" if we want to clear the bug.
On my side looking at the code I got lost. I think that the author should be able to handle it.  Is it @Aircoookie ?
Are you @blazoncek  the current maintainer of this project?

blazoncek, 2023-10-25:
> I think that the author should be able to handle it. Is it @Aircoookie ? Are you @blazoncek the current maintainer of this project?

It's me that implemented that. @Aircoookie is still the owner though.

closed 2024-04-21 by github-actions[bot]