wled/WLED #3472: Select segments in playlists

FGOD1983, 2023-10-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I want to have my ledstrip (divided in 14 uneven segments) be turned on 1 segments at a time with a certain effect and turn off again after certain time, but at the moment it is too difficult to set it up like that (if not impossible) as I would need to make a preset out of every effect I want times every segment...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

It would be way easier having the segments saved with the effects in presets and be able to select the preset in playlists and then create a delay between the segments or make it even more customizable by for example able to skip certain segments if you want.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Custom mapping/grouping, but that is almost impossible as not all segments have the same amount of LEDs...

**Additional context**
Add any other context or screenshots about the feature request here.

In my case it would make a nice transition between the steps of my staircase when the LEDs are turned on/off

Thank you for your ideas for making WLED better!

blazoncek, 2023-10-22:
Hi. Could you rephrase that? I am having difficulties understanding what you want.

FGOD1983, 2023-10-22:
As example, I have a staircase that has 14 steps and all steps have a ledstrip with data line connected to each other creating basically 1 big ledstrip. But due to some dead LEDs I have some slightly smaller strips on some steps which makes the count not consistent on all steps and I made a custom mapping so some effects can roll nicer over the steps.

I split this up in segments (each step is 1 segment) to make it easy to setup effects and colors for the stairs. 

I have several presets to setup the segments with effects/colors for themes (Halloween, Christmas, default theme during the year, ...) which I set as boot of the esp during that period of time. 

What I wanted to do is make the ledstrip turn on on motion detection but with a nice flow ( turn on segment 0 first, then segment 1 with a slight delay, then segment 2,...) And do the same when turning it off but the other way around.

The idea somebody pointed out is that this could be done with presets in playlists, but this would mean I would need to create 14 presets (for each step) per theme I have made in presets, which would be a pain in the ass...

The easiest would be that you could select a preset in playlists and see the segments in that playlist and create effects/delays between the segments in the preset or even set a segment to be ignored in a playlist. 

I hope this clears up on what I mean :)

blazoncek, 2023-10-22:
Use animated staircase usermod for that.

FGOD1983, 2023-10-22:
Oke I get that, but that brings me to another thing... Would be nice if usermods could be added after install instead of only during compile... That would make everything way more versatile as I would rather not reinstall everything at this point...

blazoncek, 2023-10-23:
Usermods are C++ code and have to be compiled alongside main code.
If you figure out a way to make them runtime installable you are welcome to implement that.
There is no such thing as "reinstall everything" with WLED. Just update OTA with correct binary (takes less than 10 clicks).

FGOD1983, 2023-10-23:
Hmmm, didn't know that they were C++ code. That explains a lot.

Also didn't know it was possible to flash the usermod binary over the normal install through OTA without losing your configuration. That is really good too know actually! Thx for letting me know!

closed 2023-10-23 by FGOD1983