wled/WLED #3473: Pixelart import saved as a preset does not work.

ShaunRoberts543, 2023-10-23:
### What happened?

If I use Pixelart import it will display image on my matrix as it should.  I can save it as a preset, but then I cannot recall that preset.  I have also tried doing a copy/past of the JSON into the preset and saving again but I get an invalid JSON syntax error.

### To Reproduce Bug

Import pixel art image, save as preset, try to recall preset.

### Expected Behavior

Preset should save.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
The JSON Pixelart is outputting for copy:

{"on":true,"bri":255,"seg":{"id":0,"i":[0,9,'ffffff','858180','231815','1f1615','271a17','291c18','241814','201615','150605','c3c1c1',18,37,'ffffff','201410','bc770d',39,41,'dd8b0b',41,48,'ffb443','e2a13e','1c1314','605d5c',51,65,'ffffff','7d7b7b','b06f0d',67,69,'dd8b0b','f8ad39',70,81,'ffb443','221715',82,95,'ffffff','92692c',96,98,'dd8b0b',98,112,'ffb443','211512',113,125,'ffffff','e38e0a','dd8b0b',127,143,'ffb443','545353',144,155,'ffffff','dd8b0b',156,173,'ffb443','a1722f',174,184,'ffffff','221713',185,204,'ffb443','231815',205,213,'ffffff','4f4644','ffb443','e5951a',216,232,'ffb443','ffbd5b','ffb443','ffbb56',235,242,'ffffff','f6f6f6','ffb945','ffb443','a76907','faae38',247,262,'ffb443','ffbe5d',263,265,'ffb443','231815',266,272,'ffffff','291d17',273,275,'ffb443','dc8903','e69926','ffb442','ffca7f',279,292,'ffb443','ffbe5d',293,295,'ffb443','f9cd91','fefefe',297,301,'ffffff','1a1612',302,305,'ffb443','dc8903','ffb443','ffba53','ffca7f',309,322,'ffb443','ffbe5d',323,326,'ffb443','221714',327,331,'ffffff','34261f',332,335,'ffb443','ee9f26','fab244',337,352,'ffb443','ffbe5d',353,356,'ffb443','977d5d',357,360,'ffffff','dad8d8','ffc169','ffca7f',363,365,'ffb443','feb341','b07d32',367,381,'ffb443','ffba54','ffbe5d',383,386,'ffb443','ffc97f','908f8e',388,390,'ffffff','1a0b06','ffb443','ffca7f',393,396,'ffb443','231815',397,410,'ffb443',410,412,'ffbe5d','ffbb54','f5a834',414,417,'ffb443','231815',418,420,'ffffff','18110f',421,426,'ffb443','774c11',427,439,'ffb443',439,442,'ffbe5d','feb342','dd8b0b',444,447,'ffb443','9b8060',448,450,'ffffff','372421',451,456,'ffb443','ffb544','45311d',458,469,'ffb443','ffbe5d','ffba55','ffb443','dd8b0b','f9ad38',474,477,'ffb443','ffd59e',478,480,'ffffff','342119',481,487,'ffb443','c37b0c','f0a22c',489,500,'ffb443','dc8904',501,503,'dd8b0b',503,507,'ffb443','ffbc5b','181818','ffffff','1a1110','ffb545',512,518,'ffb443','513512','d7870b','ffb544',521,528,'ffb443','efa023','dd8b0b','d27c06','d88509',532,538,'ffb443','211613','ffffff','0c0706','ffd49d',542,549,'ffb443','d7870b','1e1514','dd8b0b','df8d0c','ea9c24','eb9d26','e08f0b',556,558,'dd8b0b','d27d07','d07a06','dd8a08',561,568,'ffb443','140605',569,571,'ffffff','fed49d',572,579,'ffb443',579,581,'dd8b0b','c46801','201615',583,586,'c46801','cb7104','cc7404','dd8b0b',589,598,'ffb443','8f7558',599,601,'ffffff','231815','ffcf8f',603,609,'ffb443',609,611,'dd8b0b',611,613,'c76c02','1a1210','c76b02','cb7204','cf7805','dd8b0b',618,628,'ffb443','f6d19b',629,632,'ffffff','c2a47a',633,638,'ffb443','dc8a04','dd8b0b',640,643,'cc7404','624012','fed39c','d07a06','d58108','dd8b0b',648,658,'ffb443','ffd59d',659,662,'ffffff','888787','ffd69d',664,667,'ffb443','fdb240','dd8b0b',669,673,'d07a06','dd8b0b','211611','d58108','d98609','dd8b0b',678,688,'ffb443','ffd6a0','cecdcc',690,693,'ffffff','4b4847','ffd49c','ffb544','e39312',697,702,'d68308','d88509','dd8b0b','674111','ffd49c',706,708,'dd8b0b','ffb544',709,718,'ffb443','ffd6a0','655e5d',720,724,'ffffff','1d1211',725,727,'ffb443','feb441',728,735,'dd8b0b']}}
{"on":true,"bri":255,"seg":{"id":0,"i":[735,'1d1111',736,739,'dd8b0b',739,748,'ffb443','ffd6a0','0a0706',750,754,'ffffff','787070','ffd091',756,761,'ffb443',761,765,'dd8b0b','aa6b0c','ffd49c',767,770,'dd8b0b',770,778,'ffb443','ffd6a0','231815',780,785,'ffffff','291e19',786,792,'ffb443',792,796,'dd8b0b','211614','e9a235',798,800,'dd8b0b','dd8b0a',801,808,'ffb443','ffd6a0','1d1111',810,815,'ffffff','cecccc','ffb546',817,822,'ffb443','f2a530',823,826,'dd8b0b','e48f0a','ffd89f',828,832,'dd8b0b',832,838,'ffb443','ffd49c','251916',840,846,'ffffff','181211',847,852,'ffb443','dd8a08',853,856,'dd8b0b','bc760d','231815','ffd89f',859,863,'dd8b0b','e7971a',864,867,'ffb443','bd8936','1f1413','62605f',870,877,'ffffff','e0dfde','1f120e','1d1414','271c18','211717','261a15','211715','130604','d5d4d4',886,889,'ffffff','403734','231815','201615','1d1515','211715','231815','130e0d','eae9e9',897,900,'ffffff']}}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-23:
Tagging @ajotanc as the author of the functionality.

ajotanc, 2023-10-23:
When I get to work I'll check it out... From the looks of it, looking quickly, it's not getting the ID correctly.

ajotanc, 2023-10-23:
> ### What happened?
> If I use Pixelart import it will display image on my matrix as it should. I can save it as a preset, but then I cannot recall that preset. I have also tried doing a copy/past of [...]

**Just an observation... Was what you tested Pixel Magic Tool or Pixart?**

ajotanc, 2023-10-23:
I did the test here and PXM is working correctly...

[image]

ajotanc, 2023-10-23:
Use the new Pixel Magic Tool, it comes by default on WLED, http://[wled-ip]/pxmagic.htm is the most current and I'm always updating it when I have new ideas.

ajotanc, 2023-10-23:
> Thanks everyone! The Pixel Magic Tool seemed to work fine. It did not save preset through the web assignment, but I was able to set up with copy/paste JSON into a preset and that is working great. The GIF function is neat also. I appreciate the help, not sure why the PixArt would not give valid JSON for me, but this fits the bill.
> 
> Thank you again!

Do you speak PXM? If yes, everything is working fine.

It may be that the new version has not yet been released for download... I will put the updated version on my github and you can replace the version at http://[wled-ip]/edit, uploading pxmagic.htm, when If I upload the version I'll let you know!

We are here to help in any way possible! If you have ideas, improvements or whatever, just let us know and we can see what to do!

ShaunRoberts543, 2023-10-23:
Oh wait, may of spoke too soon.  Using the http://[wled-ip]/pxmagic.htm it was working fine, but then all my presets are now erased and I am unable to save any of them (including just standard WLED selections) without going to this error.
[image]
[image]

ShaunRoberts543, 2023-10-23:
I rebooted the node a couple times also to no avail.

ajotanc, 2023-10-23:
Download and replace the [pxmagic.htm](https://raw.githubusercontent.com/ajotanc/PixelMagicTool/main/pxmagic.htm) that is there with this one.

ShaunRoberts543, 2023-10-23:
Ok, so I did what you stated and installed that version to replace previous.  I get a syntax error when I "save" the image to be sent over to WLED and my presets still will not save anything.  I even did a OTA firmware update (even though I was on the newest to just freshen it up) and I still cannot save anything at all.  It will allow me to save preset, then in about a second or two I get the same window as above saying presets will not load.  Any ideas on how to clear that?  Not sure what to do now to even get the standard functionality back.

Thank you.

ajotanc, 2023-10-23:
> Ok, so I did what you stated and installed that version to replace previous. I get a syntax error when I "save" the image to be sent over to WLED and my presets still will not save anything. I even [...]

Could you send prints or send a video to better explain what is happening...?

ShaunRoberts543, 2023-10-23:
I think I got it fixed.  Something must of went haywire in the presets.json file.  I went into [IP]/edit and deleted the /.presets.json file and restarted WLED.  I then was able to save a standard preset and I have imported 2 image presets from the web tool.  All are working ok.

Is the best way to access the Pixel Magic through [IP]/pxmagic/htm, or is there something selectable within WLED?

This is a super cool program and I am geeking out over the opportunities with it.  Thank you for making it and a big thank you for the support on it.

ajotanc, 2023-10-23:
> I think I got it fixed. Something must of went haywire in the presets.json file. I went into [IP]/edit and deleted the /.presets.json file and restarted WLED. I then was able to save a standard [...]

Good, I'm glad you managed to solve it... In the new update there will be an access button for it, this @blazoncek can answer you with more certainty as to when it will be...

Below is an example of how it works...

https://github.com/Aircoookie/WLED/assets/47322034/c2b08b45-187d-4356-87b6-9dc18c046c89

blazoncek, 2023-10-23:
@ajotanc I am not seeing pixelart logo on embeded version.
[image]

ajotanc, 2023-10-23:
> @ajotanc I am not seeing pixel magic logo on embeded version. [image]

Are you talking about the last one I went up?

ShaunRoberts543, 2023-10-24:
Last question, I PROMISE!  What are the abilities/parameters of using a GIF.  File type, name, size constraints?  I am not getting many to work and I cannot seem to find the common issue as to why.

Thanks!

blazoncek, 2023-10-24:
@ajotanc the minimised version embedded in WLED firmware lacks image. Apparently the `<img ...>` tag is incorrect.
I've updated it in 0_15 branch so please use that for your further enhancements to prevent merge conflicts.

ajotanc, 2023-10-24:
> Last question, I PROMISE! What are the abilities/parameters of using a GIF. File type, name, size constraints? I am not getting many to work and I cannot seem to find the common issue as to why.
> 
> Thanks!

By the end of the day I will try to send a video as an example.

ajotanc, 2023-10-24:
> @ajotanc the minimised version embedded in WLED firmware lacks image. Apparently the `<img ...>` tag is incorrect. I've updated it in 0_15 branch so please use that for your further enhancements to prevent merge conflicts.

I updated the PR, but it is closed, I don't know if you can see the change.

blazoncek, 2023-10-24:
@ajotanc once the PR is merged you need to make a new one.
Please use 0_15 branch.

ajotanc, 2023-10-24:
> Last question, I PROMISE! What are the abilities/parameters of using a GIF. File type, name, size constraints? I am not getting many to work and I cannot seem to find the common issue as to why.
> 
> Thanks!

I made a new correction, it was actually breaking the animation, but now it's working... I'll update it on my GH and you'll update it again.

**New version**
[pxmagic.htm](https://raw.githubusercontent.com/ajotanc/PixelMagicTool/main/pxmagic.htm)

ajotanc, 2023-10-24:
> Last question, I PROMISE! What are the abilities/parameters of using a GIF. File type, name, size constraints? I am not getting many to work and I cannot seem to find the common issue as to why.
> 
> Thanks!

Example of animation working in wled.
https://github.com/Aircoookie/WLED/assets/47322034/7c3ff31e-b6e8-4347-9d8c-cfb2ce4262ba

ShaunRoberts543, 2023-10-24:
This is so fun!  Do you think there is any chance of the P4(?) matrix panels being supported at some point?  They are a lot cheaper, but I do not know much about them.

Here are a couple of examples of what I did:

https://github.com/Aircoookie/WLED/assets/84977157/8290d620-0587-49f3-b0b0-8e3817eab7dd

[image]

ajotanc, 2023-10-25:
> This is so fun! Do you think there is any chance of the P4(?) matrix panels being supported at some point? They are a lot cheaper, but I do not know much about them.
> 
> Here are a couple of examples of what I did:

Too cool, hahaha! I'm glad you liked it. So, here to say if this model is supported is @blazoncek, he responds and has the same sequence in the json, I don't see a problem with it working if it supports it.

blazoncek, 2023-10-25:
Please use WLED [forum](https://wled.discourse.group) or Discord for chat.
And close the issue if the problem is solved.

closed 2024-01-02 by blazoncek