wled/WLED #3474: Chase 2 not applying across the whole strip

jgallen23, 2023-10-23:
### What happened?

I have a set of fairy pixels that are 100 leds. I confirmed that the segment is set to 100, but the animation stops half way through. Looking at peak as well confirmed this. I also tried blink and it also only works on the first half of the light strand, but I can definitely control the color of the 2nd half of the strand. I also tried this on my other two fairy light strips (200 and 100 leds) and the same thing is happening, but I tried on my 60 led light strip and it animates properly. I just noticed this when updating to v14. When I do something like Lake, though, the entire strip animates correctly. 

[image]

This is the light strip I am using: https://www.amazon.com/gp/product/B0CCRZX9B8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### To Reproduce Bug

- Go to effects and chose Chase 2
- Look at Peak to see that the animation stops half way through

### Expected Behavior

Animation should continue throughout the entire strand

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jgallen23, 2023-10-27:
Figured it out. On the colors screen, it was set to colors 1&2 so it was using the bg color for the second half

closed 2023-10-27 by jgallen23

CDeLeon94, 2024-12-12:
I'm not sure how long it would have taken me to figure that out, I wanted chase to have those two colors, "colors 1&2" was set to the right 2 colors. 

Thanks to your comment, I set the palette to "color 1" and everything works as intended