wled/WLED #3477: Led Voltage setting to 12V isn't saved for "Recommended power supply for brightest white:"

Maximus48p, 2023-10-23:
### What happened?

Led voltage changed from default 5V to 12V. Recommended power supply for brightest white: is also changed to 12V which is good.
But after saving the "Recommended power supply for brightest white:" is back to 5V. Led Voltage is correctly saved to 12V, see screendump.

Used on:
Athom LED Strip controller
Model: LS-4P
Input Voltage: DC 5-24
Output current: 16A
wled binary file: WLED_0.14.0_ESP02.bin.gz
[image]

### To Reproduce Bug

change back and forth Led voltage setting.

### Expected Behavior

After saving the "Recommended power supply for brightest white:" should correspondent with the Led Voltage.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0_ESP02.bin.gz

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-23:
I cannot reproduce. On my system saving any value persists.

dmaclach, 2024-01-17:
I am seeing the same problem. QuinLeg-Dig-Quad.

[image]

blazoncek, 2024-01-17:
The current calculation is correct. The only issue is that it displays 5V instead of 12V.
The voltage is removed in 0.15 (as it really has no meaning).

It may get fixed in 0.14.2 if it is ever released but do not hold your breath.

netmindz, 2025-01-10:
Can we close this now then @blazoncek ?

blazoncek, 2025-01-10:
IMO yes.

closed 2025-11-08 by netmindz