wled/WLED #3482: API Request (Philips Hue Polling Light) is not Working for me.

itwegner72, 2023-10-24:
### What happened?

The API Request

&HP=  0 to 99 (Sets Hue polling light ID (0 is off) | 0.5.1)

Is not working for me. Nothing was changed ?

MFG
Ingo

### To Reproduce Bug

curl "http://192.168.178.181/win&HP=0"   for off (no Sync)
curl "http://192.168.178.181/win&HP=26" polling the Light 26 (Sync)

### Expected Behavior

I want to switch off and on the Hue Sync between my Hue Light and the Wled Instance.

### Install Method

Binary from WLED.me

### What version of WLED?

latest

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-24:
Nothing was changed in HUE API. Make sure you have HUE enabled WLED binary as it is optional compile.

itwegner72, 2023-10-26:
Test it by yourself, is not working.

blazoncek, 2023-10-26:
> Test it by yourself, is not working.

I will when you send me Hue lights. 😄 Unfortunately I do not own Hue or compatible lights.
Still, Hue code in WLED was not changed in last 2 years (there was a whitespace cleanup 9 months ago, but only whitespace). I would assume something wrong with your set-up or environment.

itwegner72, 2023-10-26:
Have you testest it, No , Or ?

itwegner72, 2023-10-26:
[image]
[image]

itwegner72, 2023-10-26:
[image]

Nothing was changed

closed 2024-04-21 by github-actions[bot]