wled/WLED #3485: Playlist transition time not working

marcobrianza, 2023-10-25:
### What happened?

The transition time between preset in the playlist editor is not applied as in 13.3

### To Reproduce Bug

set a transition time of 5 seconds, you see a transition that starts with the right timing but quickly (after 0.7?) it jumps the next playlist entry

### Expected Behavior

The transition time of the playlist editor should be the time between playlist presets

### Install Method

Binary from WLED.me

### What version of WLED?

14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-25:
Please post presets used and playlist.

marcobrianza, 2023-10-25:
[wled_presets.json](https://github.com/Aircoookie/WLED/files/13170103/wled_presets.json)
[wled_cfg.json](https://github.com/Aircoookie/WLED/files/13170105/wled_cfg.json)

marcobrianza, 2023-10-26:
I have discovered that the software uses as transition time the one in the preset (example "transition":50) and not the one in the playlist editor.

See attached modified presets where I set 5s in the preset and 1s in the playlist

[wled_presets-mod.json](https://github.com/Aircoookie/WLED/files/13180357/wled_presets-mod.json)

blazoncek, 2023-10-26:
> I have discovered that the software uses as transition time the one in the preset (example "transition":50) and not the one in the playlist editor.

This could be a bug. Will need to investigate.

blazoncek, 2023-10-26:
Tested on my ESP32 and it works as expected. Setting transition time in playlist has to be shorter or equal than effect duration time.

marcobrianza, 2023-11-01:
I am using an esp8266 2M controller because is a product. Do you know a ready made esp32 controller with pwm output so I can buy and test?

blazoncek, 2023-11-01:
Please check [KB](https://kno.wled.ge/basics/compatible-controllers/) for recommended ones.

closed 2024-04-21 by github-actions[bot]