wled/WLED #3486: new PWM RGB+WW+CW mode

marcobrianza, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
With RGB+CCT both the white channels cannot be set to max intensity at the same time.
There is an issue in transitions with this mode.

**Describe the solution you'd like**
I would like a 5 PWM preset where the WarmWite and ColdWhite can be set with a dedicated slider. 

**Describe alternatives you've considered**
a possible solution is to use an RGB and 2 separate PWM channels. This maxes out all the available channels on a esp8266 and I think the interface is not consistent and easy to use.

**Additional context**
I would use this mode in setups where I have a RGB+W+UltraVilolet

Thank you for your ideas for making WLED better!

blazoncek, 2023-10-26:
Did you set CCT blending to 100 in LED settings?

marcobrianza, 2023-11-01:
Hi, thank you for pointing to the CCT blending feature. It fixes the intensity issue. I still think that the feature is useful for the transition issue and for handling 5 channels independently.