wled/WLED #3487: Messing up some 1d effects in 2d matrix sync received

DarthWeber, 2023-10-26:
### What happened?

e.g. Bouncing ball works fine also on 2d matrix setups. But when synced to another matrix, the slave changes to bouncing ball too, but it looks messed up like the orientation of the matrix is suddenly destroyed or something like that. To reproduce:

Both instances has no problem in presenting bouncing balls independently without sync.

The effects like meatballs doesn't show this issue.

### To Reproduce Bug

Master wled 14.0 with sync bouncing balls on a matrix 16wh60
Slave wled 14.0 also matrix, but different dimensions 6w60h and other orientaton

### Expected Behavior

Correct sync of 1d effects on 2d effects ins sync mode

### Install Method

Self-Compiled

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
No one
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-26:
Synced how? UDP sync? DDP?

DarthWeber, 2023-10-26:
By UDP

blazoncek, 2023-10-26:
UDP sync (even with options enabled) will not sync every segment option, including sound simulation and 1D-2D mapping.
1D-2D mapping is required for correct 1D effect behaviour.

This is a shortcoming of UDP sync as only 1 byte is reserved for segment options (whereas it requires full 16 bit).

closed 2024-01-04 by blazoncek