wled/WLED #3488: Rolling Text

Diramu, 2023-10-26:
I discovered that is the in the "scrolling text" effect, completely fits in the array, it does not scroll.

Not a big deal, I can just add  blanks, or even repeat the message to make it long enough to scroll.

Idea: But giving a static message, how about an option (or an new effect) where this text rolls (scrolls vertically).

It possible that both, scrolling and rolling at the same time, might be a nice effect.

blazoncek, 2023-10-26:
This is already implemented in 0.14

closed 2023-10-26 by blazoncek

Diramu, 2023-10-26:
I could not see a feature to make the text roll.  Maybe I missed it.
I found the transpose box, that rotated the text orientation by 90 degrees, scrolled it vertically. 
That not want I looking for. I want the movement this that direction, without the orientation of the text to change.

A normal scroll (left to right),  data from column N, is replaced with data from column N-1, and new data (or column MaxX) enters column 0.
I'm hoping for a roll (top to bottom), row N, is replaced with row N-1, and new data (or row  MaxY) enters row 0.