wled/WLED #3491: ESP32-S3 only outputs to 4 strips

thirstyice, 2023-10-29:
### What happened?

The [docs](https://kno.wled.ge/features/multi-strip/) claim that the ESP32 can support 10 strips output, but when I add more than 4 strips I can only control the first 4. This seems to be independent  of which GPIO are used by each of the strips.
If turn on "Make a segment for each output" then the segments are created as expected, but when trying to control any other than the first 4, the interface removes the color wheel.

### To Reproduce Bug

Wire an ESP32-S3 (in my case an [M5Stamp S3](https://docs.m5stack.com/en/core/M5StampS3%20PIN2.54)) with >4 strips of output.
Observe that you only have control over the first 4 strips

### Expected Behavior

Control of up to 10 strips of LEDs

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0 build 2310130

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-29:
There is a difference between regular ESP32 and ESP32-S3. They are completely different chips.
S3 can only output 4 digital and 2 analog RGBW signals.

closed 2023-10-29 by blazoncek

thirstyice, 2023-10-29:
Ok, thanks. May be worth clarifying the documentation on that.

blazoncek, 2023-10-29:
Please consult Espressif documentation.
S2, S3 and C3 are still experimental as far as WLED goes.

thirstyice, 2023-10-30:
Which part of the espressif documentation?
I just need to know what I'm looking for, I should be able to figure it out from there.

blazoncek, 2023-10-30:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.