wled/WLED #3492: Soap effect crashes esp8266

PKCubed, 2023-10-31:
### What happened?

I set up a matrix on an esp8266 from a single pin. Whenever I click on the "soap" effect the esp8266 will restart and the effect does not appear.

### To Reproduce Bug

Here's what I did. I have an esp8266 running an led strip in a serpentine pattern and set up with the 2D configuration option. All effects work perfectly fine except the "soap" effect. In order to get the board to crash, all I did was click on the "soap" effect.

### Expected Behavior

The soap effect should have displayed an actual effect instead of rebooting the board.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-10-31:
It works well on my set-ups. Please post your configuration.

savdagod, 2023-11-05:
Same for me on an ESP8266 with a 2D matrix configured

blazoncek, 2024-03-04:
No feedback provided and not reproducible in my environment.

closed 2024-03-04 by blazoncek