wled/WLED #3493: FR: Input field for parallel string count to make current limiter work with parallel strings.

mplinuxgeek, 2023-10-31:
**Is your feature request related to a problem? Please describe.**
Current limiter is hard to use when using parallel strings/strips, for example I modified a cheap star from China which uses 12 strings of 9 WS281x compatible LEDs connected in parallel, works perfectly fine (with a level shifter) but getting the current limiter working requires some fudging.

**Describe the solution you'd like**
Add a input field to set the number of parallel strings that will be used in the current limit calculation.

**Describe alternatives you've considered**
I've set the LED voltage/current to custom and used the maximum value of 254mA, 12x55mA should be 660mA but custom doesn't allow higher than 254mA, I've then set the current limiter to 550mA which gives me a max current usage of about 900mA which was my target to power this from a USB charger.

**Additional context**
I understand this is a niche use case but it would be very helpful when using parallel strings.

Thank you for your ideas for making WLED better!

blazoncek, 2023-10-31:
Just multiply per-LED current consumption with number of strips and enter custom value.

mplinuxgeek, 2023-10-31:
> Just multiply per-LED current consumption with number of strips and enter custom value.

This is exactly what I tried and as per the alternatives comment, but it doesn't work for the numbers I need, I need the custom number to be 660mA but the field accepts a max of 255 as per an error message however doesn't actually accept 255, it does accept 254 however.

mplinuxgeek, 2023-11-01:
> Well... You're stuck, then. Unless someone tackles this.

That's why I'm submitting a feature request, if and when I get some time I can have a go at hacking it in myself and submitting a PR but I'm not a developer by any measure.