wled/WLED #3494: Audio Reactive profiles wont show up

pranjalchanda08, 2023-10-31:
### What happened?

Audio reactive profiles does not show up when LED setting is set to PWM-RGB.
The same is not seen even after falling back to WLED mode.
Default the profiles are visible.

Had to factory restore/reload firmware to get back the profiles.

### To Reproduce Bug

1. Set the LED to PWM-RGB
2. Reset the LED to WLED

### Expected Behavior

The profiles shall be visible.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pranjalchanda08, 2023-10-31:
After doing some tweeks I understand that this happens when total number of LED set is only 1.

blazoncek, 2023-10-31:
If by "profiles" you mean effects (or sometimes called modes). That's by design.

pranjalchanda08, 2023-10-31:
True. This can be documented somewhere or shown as a warning in the UI for people to understand their fault. You may use this issue to implement that caution waring or document and close this issue.

blazoncek, 2023-11-01:
You are welcome to update the documentation: [WLED-docs](https://github.com/Aircoookie/WLED-docs)

closed 2024-04-21 by github-actions[bot]