wled/WLED #3502: Dissolve/Dissolve Rnd with secondary color stopped working in 0.14

TripleWhy, 2023-11-05:
### What happened?

Dissolve doesn't do much when there is a background color set isn't black:
It will dissolve to to the background color keep all LEDs at the background color indefinitely.

### To Reproduce Bug

0. Use WLED 0.14.
1. Select the Dissolve (or Dissolve Rnd) effect.
2. Chose a background color that isn't black.

### Expected Behavior

Fills LEDs with primary in random order, then with background color, then repeats.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

This worked in WLED 0.13.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-05:
It works as expected on my system.

blazoncek, 2023-11-06:
Nope. Works on ESP32, ESP32-S2, SEP32-C3 and ESP8266.
Matrix or not. Brightness 200 or less, palette Default or any other. Fx and Bg any color. Random or not.

Must be something particular to your set-up. Have you tried "Global LED buffer"?

TripleWhy, 2023-11-06:
Enabling that option does indeed fix this.

When I copy a configuration backup from my ESP8266 to my ESP32, I have to enable the option as well, otherwise I get the same issue.  
However, when I erase the ESP32, re-flash the firmware, and try with a clean setup, I don't need the option. Strange, but I don't know if you consider that an issue or not.

blazoncek, 2023-11-08:
The problem lies in this comparison: `SEGMENT.getPixelColor(i) == SEGCOLOR(1)` which may never be true if the global brightness (or even segment opacity) is not 255.

blazoncek, 2023-11-08:
I think I have it sorted out.

closed 2024-01-04 by blazoncek