wled/WLED #3504: Could a new controller be added to the Ethernet dropdown?

jasonsomers, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
No

I manufacture controllers that run WLED. My new controller will have an option for a pluggable Ethernet module.
I would love it if you could add this to the list of supported Ethernet modules in the Ethernet version of WLED.
Please just refer to it as RGB2Go Tetra

  // RGB2Go Tetra
  {
    1,                    // eth_address, 
    -1,                   // eth_power, 
    23,                   // eth_mdc, 
    18,                   // eth_mdio, 
    ETH_PHY_LAN8720,      // eth_type,
    ETH_CLOCK_GPIO17_OUT  // eth_clk_mode
  },

I am including some pics of the board and module in case you are curious about it.
[image]
[image]

Thank you!!!
-Jason 
RGB2Go

blazoncek, 2023-11-06:
Just make a PR. And update [documentation](https://github.com/Aircoookie/WLED-Docs).

jasonsomers, 2023-11-06:
Blaž,
Unfortunately, I am not a software developer.
Is there any assistance you could provide?
Thanks,
Jason

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

blazoncek, 2023-11-06:
Can you contact me on Discord?

blazoncek, 2023-11-06:
It appears that your Ethernet configuration is the same as `ESP3DEUXQuattro` AKA `ESP3DEUX`

blazoncek, 2023-11-06:
You may just want to update [WLED docs](https://github.com/Aircoookie/WLED-Docs) to include information about your hardware.

softhack007, 2023-11-11:
> You may just want to update [WLED docs](https://github.com/Aircoookie/WLED-Docs) to include information about your hardware.

@jasonsomers does your board also work with existing settings, as proposed by @blazoncek ? If yes, then it would be preferable (imho) to just update the documentation.

blazoncek, 2023-11-11:
@softhack007 speaking of that, there are several Ethernet controllers with same settings (their only difference is their name) so IMO we should condense all of those into a single entry.

softhack007, 2023-11-12:
> @softhack007 speaking of that, there are several Ethernet controllers with same settings (their only difference is their name) so IMO we should condense all of those into a single entry.

@blazoncek good point 👍
We should avoid duplicate config array elements, to save some RAM and PROGMEM. 

Maybe it's best to de-couple the ethernet config array from UI selection, so users could still select the ethernet board as before, but then several selections would be mapped to the same config. Maybe that's something for 0.14.2 or 0.15.0?

jasonsomers, 2023-11-12:
Why not just have us enter in the pins, like in other add-ons?
That would be fine with me as long as I could roll an image where I can set
the defaults for my board.
-Jason

blazoncek, 2024-02-09:
49ceac1ac724b6c7537f8919876947c848014334

closed 2024-02-09 by blazoncek