wled/WLED #3505: USERMOD_BATTERY crashes Audioreactive with analog mic MAX9814

fatecasino, 2023-11-06:
### What happened?

I have set up a portable wled mini system using Lipo battery. In order to activate the audioreactive mode I have added a MAX9814 mic module.
The main problem is that after a few minutes the audioreactive stops to work, whereas the non music effects still play flawlessly. 
I did plenty of tests and it ended to be that not all usermod go well together and in my case after adding a couple of them I ended up not knowing what’s wrong. The following usermods from aircookie repository have been added to my ESP32 system:

- Display ALT
- Rotary Encoded ALT
- Autosave
- Battery

I started to install one by one to see what’s wrong and strangely but definitely confirmed the Battery usermod makes the audioreactive to crash after a few minutes of playing. If I remove the battery usermod the audioreactive plays without problem for hours and hours (i have tested this). When I include the battery usermod in the build option and upload it the audioreactive stops after 4-5 minutes.

My build option for battery is:
-D USERMOD_BATTERY -D USERMOD_BATTERY_USE_LIPO -D USERMOD_BATTERY_TOTAL_CAPACITY=1200 -D USERMOD_BATTERY_LOW_POWER_INDICATOR_ENABLED -D USERMOD_BATTERY_LOW_POWER_INDICATOR_THRESHOLD=10 -D USERMOD_BATTERY_MEASUREMENT_PIN=34

It is a Firebeetle 2 ESP32 which has an integrated LiPo charger which is internally connected to GPIO 34.

Is there a get around for this issue?

### To Reproduce Bug

Install the latest wled
install Battery usermod
install MAX9814 to analog input A0

### Expected Behavior

All audioreactive effects stop playing. The non music effects keep playing flawlessly. In order to recover the audioreactive mode I have to completely power off and on again the ESP32

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-06:
Reading analog inputs is incompatible with Audioreactive usermod.
Not a bug.

Please use I2S microphone.

fatecasino, 2023-11-06:
Sorry, I was not aware of this prerequisite.