wled/WLED #3510: Add Support for SM16825 and NeoPixelBus 2.7.7

charlievogt, 2023-11-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

LED manufacturers are coming out with devices using the SM16825 chip.  WLED does not currently support it, but NeoPixelBus' latest release does.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I'd like for WLED to add support for the new chip and NeoPixelBus' updated library.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I'm able to control the LEDs with an ESP32 and Arduino sketches, but I would really like to be able to use WLED instead of needing to write every effect.

**Additional context**
Add any other context or screenshots about the feature request here.

Reddit post about it:  https://www.reddit.com/r/WLED/s/pEaNBrGAks

Updated NeoPixelBus feature:  https://github.com/Makuna/NeoPixelBus/issues/740

Thank you for your ideas for making WLED better!

SkHCrusher, 2023-12-04:
You talk about „Govee Outdoor Permanent Lights Pro“. I have the „Govee Outdoor Permanent Lights“ (without pro) and want to control them.
Is this the same codec and issue? :-)

T3rry12, 2023-12-14:
I would also love this, i just boguht a 200ft kit of the pro lights and only need just over half of it, so it would be wonderful to be able to use them up with WLED controllers, also would probably just switch the main set to WLED when it's supported.

kennygay, 2024-02-01:
I'd love WLED support for SM16825 specifically for my Govee Pro installs

tungmeister, 2024-02-18:
I'm looking to purchase a set of these and it would be great if I could control them via WLED. Forgive my ignorance but why is this targeting 16 instead of 15? what's involved in adding support, is it not just updating the library to =<2.7.7 and adding some definitions? I'd be happy to have a go and get these up and running once I've purchased if someone could give me some pointers on what's required from a coding standpoint.

mworsnop, 2024-04-03:
I would also like to have WLED control the Govee pro outdoor lights.

rmungeer, 2024-04-08:
Please do add support

T3rry12, 2024-08-05:
is there anythign we can do to push this? seems like all the actual work is done, just needs put in to the app... i have a ton of the pro lights and my controller died so i would love to make use of them.

jaggie09, 2024-08-05:
I don't know if this is appropriate, but I'd be willing to donate a couple hundred dollars to help this get done.  I've got quite a few strings of these and would really love to get off the govee the controllers.

T3rry12, 2024-08-05:
agreed, i dont know if i have that much money, but i could make a small donation

adamskoreyko, 2024-08-05:
Again, not sure this is allowed but I would contribute as well.

T3rry12, 2024-08-05:
hey all, i asked on the wled discord, and they said if you want to donate you can do so via https://www.paypal.com/paypalme/blazoncek this is the main dev, and you should be able to add notes within the donation that you want to see this issue escalated.
[image]

jaggie09, 2024-08-05:
Donated $300 USD for the cause.  Understand there are more pressing matters and really appreciate your hard work!  

Would love to have this by Holiday season to help put these lights into my Holiday display.

blazoncek, 2024-08-05:
That was ridiculous! 😮
@jaggie09 contact me on Discord, please.

M0LTE, 2024-08-05:
Donated what I can.

T3rry12, 2024-08-05:
guys, if you donated, please contact blazoncek on discord, he provided me an image and i tested it on 8266 and it works great https://photos.app.goo.gl/kaXZKDtGLMSTAFh48

mworsnop, 2024-08-05:
I would love to donate. I was waiting to hear how we're supposed to do
that. Also, what is the way of contacting this person on discord I would be
happy to do that too, but I don't know exactly where on discord I would
find them. Sorry I'm just a stupid dumb user, I know a little bit about
flashing 8266 but that's about it.

https://photos.app.goo.gl/kaXZKDtGLMSTAFh48

T3rry12, 2024-08-05:
i commented a few posts above with the info on where to donate, as for contacting on discord you should be able to do that by making a new private message in discord directed at his username, also shared above.

jaggie09, 2024-08-06:
> guys, if you donated, please contact blazoncek on discord, he provided me an image for testing and i tested it on 8266 and it works great so far https://photos.app.goo.gl/kaXZKDtGLMSTAFh48

What settings did you use?  I can't seem to get mine to work correctly on the SM16825 RGBCW on the ESP32 board I have.

mworsnop, 2024-08-06:
I sent a $200 donation to him. I’m guessing he’s probably asleep because
he’s on Europe time somewhere. Very anxious to check it out that sounds
wonderful

T3rry12, 2024-08-06:
> What settings did you use? I can't seem to get mine to work correctly on the SM16825 RGBCW on the ESP32 board I have.

i picked the sm16825 option on the led preferences settings and that was it, idk if you have the wiring fired but pink is positive, blue is negative, yellow is data, and make sure you have the direction the right way

T3rry12, 2024-08-06:
> I sent a $200 donation to him. I’m guessing he’s probably asleep because he’s on Europe time somewhere. Very anxious to check it out that sounds wonderful
> […](#)

yes, he said to me he was going to bed about 3 hrs ago

jaggie09, 2024-08-06:
> > > guys, if you donated, please contact blazoncek on discord, he provided me an image for testing and i tested it on 8266 and it works great so far https://photos.app.goo.gl/kaXZKDtGLMSTAFh48
> > 
> > 
> > What settings did you use? I can't seem to get mine to work correctly on the SM16825 RGBCW on the ESP32 board I have.
> 
> i picked the sm16825 option on the led preferences settings and that was it, idk if you have the wiring figured but pink is positive, blue is negative, yellow is data, and make sure you have the direction the right way, and make sure you combine the negatives.. i was using a 8266 board, didnt have any 32's on hand

I think I got it figured out.  I noticed when I first started WLED, Output 1 was stuck on WS281x (drop down wouldn't let me change) with GPIO at 16 so I just tried making the string length as 1 (smallest it would let me).  I then used Output 2 as SM16825 with GPIO at 2.  I was getting bad strobing with this even with the Output 1 segment isolated and off. 

Once I figured out that deleting Output 2 would let me change output 1 to SM16825 things cleared up.  Only other thing I had to do with flip CW and WW which if I remember correctly was discussed on the NeoPixelBus side of things.

So far so good!  Thanks for your help Blazoncek!

blazoncek, 2024-08-06:
I am overwhelmed! You all are crazy. 😄 
I've sent binaries on Discord you can share them with others if you want but please provide feedback. Anything.
Here or on Discord beta channel.
Once you all agree that the code works well (the SM16825 part) it will be included in stock WLED.

RudeBoiG, 2024-08-08:
Does anyone know if Eufy E22 uses the same SM16825 chip. I picked up a set and would like to test and see if works with that set. Thank for the great work.

RudeBoiG, 2024-08-08:
Tested...the Eufy E22 Permenant lights uses WS2805 RGBCW instead SM16825.

T3rry12, 2024-08-08:
> Tested...the Eufy E22 Permenant lights uses WS2805 RGBCW instead SM16825.

lucky you, it's already supported 😊

lightingman117, 2024-08-09:
> I am overwhelmed! You all are crazy. 😄 I've sent binaries on Discord you can share them with others if you want but please provide feedback. Anything. Here or on Discord beta channel. Once you all agree that the code works well (the SM16825 part) it will be included in stock WLED.

You told a bunch of bougie Govee Pro owners that money can solve their problem. What did you expect ;)
Haha, happy to contribute to progress!

Tested working, 90 pixels; Govee Pro (SM16825); Needs 3.3>5.0V level shifter (MFG says 28800bps).

blazoncek, 2024-08-09:
> You told a bunch of bougie Govee Pro owners that money can solve their problem.

Did I? I do not recall such promise. 😄 

> What did you expect

A beer and not much more. But that was not my intent. There was a guy from Australia who sent me his LEDs on his own expense to get support for them. When I see dedication like that (from a single person) I do have *incentive* to help (if I can).
And BTW that work paved the way for SM16825 to be fairly smooth upgrade.

M0LTE, 2024-08-16:
Turns out I have the non-pro lights. I could have been using WLED all this time. So no feedback from me. Thanks though!

t0nyz0, 2024-08-25:
I had a few extra from a home install then I found this post You guys are awesome. Its working flawlessly, much appreciated!!!!

[image]

[image]

T3rry12, 2024-08-25:
> I have a few extra lights from my home install, then I found this post. You guys are awesome. Its working flawlessly, much appreciated!!!!

I did similar, I printed a holder for my daughter's vanity to hold 5 of them, got a d1 mini and a 5v step down in there
[image]

closed 2024-10-27 by blazoncek

newdamage1, 2024-11-30:
> I have a few extra lights from my home install, then I found this post. You guys are awesome. Its working flawlessly, much appreciated!!!! [...]

What hardware are you using for this setup? I've been tinkering with my extras with an esp8266 and an amazon cheapy level shifter, and I don't get anything more than flickering. Suggestions would be greatly appreciated!

balucanb, 2024-12-18:
So happy to see the Govee pros being supported! Thanks @blazoncek !!! Just purchased 200 ft of these and was under the impression I could not use WLED, so very happy I was wrong.

blazoncek, 2025-06-27:
Please use [Discord](https://discord.gg/FMvz8nzmat) or WLED [forum](https://wled.discourse.group) for help and support questions.