wled/WLED #3517: Manila time zone (UTC+8) is not available in the "Time Setup" page, Time Zone field.

Lutier2017, 2023-11-11:
### What happened?

I want to set the Manila time zone (UTC+8), but it is not available in the "Time Setup" page, Time Zone field.

### To Reproduce Bug

Open the "Time Setup" page, and it will not be possible to set Manila (UTC+8) Time Zone.

### Expected Behavior

when I open Open the "Time Setup" page, and check the list of available time zone, Manila or (UT+8) should appear to be selected.

### Install Method

Binary from WLED.me

### What version of WLED?

V0.14.0 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-C3

### Anything else?

Website: https://kno.wled.ge/features/settings/#time-settings
Indicates that I should open this request to include this new "Time Zone".

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-04:
@Lutier2017 to add the Manila timezone, we need to know if - and how - daylight savings time is applied in your area.
* do you have DST during summer?
* if yes, when are your clocks "moving forward"? When are they set back?

Lutier2017, 2023-12-06:
Hi Frank,

Thanks for your email.

The answer sto your questions is as follow:

1) do you have DST during summer?
   We do not have DST during summer in the Philippines.

2) if yes, when are your clocks "moving forward"? When are they set back?
    We do not have DST during summer in the Philippines.

Please, acknowledge receipt of this email.

Sincerely Yours,

Claudio Passarella

softhack007, 2023-12-14:
Hi, it looks like you can use the "China" timezone, which is also UTC + 8 hours, no daylight saving time.
Please select "CST(AWST)" in timezone settings. As we don't have aliases for other time zones in our list, this means we would just need to append _PHST_ to the brackets after CST.

closed 2023-12-30 by softhack007