wled/WLED #3518: Preset JSON api freeze behavior not as expected

mechalv, 2023-11-12:
### What happened?

I was trying to have a preset frozen, so I changed the present value `"frz"` from `false` to `true` manually. After clicking on "Apply" the result has been as intended. Then I have pressed on another preset to test another the effect, and later on I have tested the frozen preset again, but I have found out that it was no longer working as it should. Now it freezes the effect/preset that is coming from, not the preset that `"frz":true` is written on.
Feels as if WLED applies the segment configuration before the LED light output, so it freezes before changing to the desired effect. The workaround would be splitting one preset into 2 and have a playlist, to first change to the effect and then immediately freeze it with a preset such as `"X":{"n":"X-Freeze","seg":[{"frz":true}]}`.

### To Reproduce Bug

1. Create a preset with any moving effect and save it
2. Then edit the preset and change `"frz":false` to `"frz":true`, and save it
3. Click on another (different) moving effect or preset
4. Click again on the frozen saved preset from step 2
5. See that the effect that freezes is the one from step 3 and not step 1 (as desired)

### Expected Behavior

Freeze the effect that is written on the preset at the same time of `"frz":true` please. Thank you.

### Install Method

Self-Compiled

### What version of WLED?

0.14.0 build 2310130 (QuinLED bin)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-12:
Not a bug. Freeze segment works as intended.

As segment is part of a preset its state is determined by preset settings, which include freeze state. Furthermore some actions will unfreeze segment on their own.

Consider freeze as a temporary segment state (also used for PixelArt display or individual pixel control).

closed 2023-11-12 by blazoncek

mechalv, 2023-11-12:
> Not a bug. Freeze segment works as intended.
>
> As segment is part of a preset its state is determined by preset settings, which include freeze state. Furthermore some actions will unfreeze segment on [...]

I don't know if I understand you correctly. The `"frz"` option is there, but users are not expected to use as I intend to use it because it is used for other things??
What should I do to have an effect to pause then? Thanks

blazoncek, 2023-11-12:
Effect cannot be paused. Freeze only stops updating LEDs on the segment it does not stop effect from running.

blazoncek, 2023-11-12:
I have to correct myself. Some effects are indeed paused while the segment is frozen. But that is due to nature of the effect, not all behave that way. Time still passes on.

In general, if the effect is time based, unfreezing a segment will not continue from the point it was frozen.