wled/WLED #3519: Add an optional current limiter per segment

larrywal, 2023-11-12:
**Problem:**
Based on how I do my wiring and injection, certain segments can handle more power than others.  Rather than having a single max current as a backstop to a safety issue, I'd like the ability to apply a current limiter per segment.   As an example, my 350W meanwell PSU @ 12V can handle let's conservatively say 20A.  So I hook that to my DigQuad, but for a 400px prop where I want to power balance since I'm only going to run the prop at 30% brightness, then I need to current limit the prop.

**Proposed Solution**
Allow me to add a max current per segment.

**Describe alternatives you've considered**
There is no alternative safety backup short of wiring things differently.

**Additional context**
I made this request on the discord and several other folks wanted this too.

softhack007, 2023-11-12:
Hi,
This feature has been proposed several times already in the past, however it is not easy to implement. It cannot be per segment (segments are volatile and get deleted/created whenever you load a preset). However it might be possible - in principle - to have current limits per LED PIN (aka "bus"). 

But this will need a major rewrite of the way we integrated the NeoPixelBus drivers in WLED. It could be a candidate for 0.15. next year, if someone volunteers to write the code.

blazoncek, 2023-11-13:
The problem of current limiter is multi-fold.

First, `strip` AKA `WS2812FXˇ` class, where ABL resides, has no knowledge of physical LED layout (multiple pins used, LED gaps, etc)
Second, correct GPIO/output ABL implementation would require adding single LED max current to every GPIO/output used i.e. every strip created in LED settings. It would also require to add power budget to each GPIO/output. This would cause existing set-ups/installations to no longer be valid.
Third, it would require ABL code to be moved from `WS2812FX` class into `DigitalBus` class, but `DigitalBus` class has no knowledge of how segments are created, possibly producing uneven brightness of segments.
Fourth, current implementation restores brightness to its original value after pixels are painted, this may be impossible on the bus level, producing unwanted results in `getPixelColor()` method.
Fifth, segments may have gaps when using ledmaps of gap file which should be excluded from ABL.
Perhaps there are more problems still.

Adding additional fields to store LED current and power budget to each GPIO/output may not be easy on ESP8266 as it already struggles with available RAM and I would dislike such feature disparity between it and ESP32.

All that said, I am not against it. It is just that implementing correctly will take time and resources.

larrywal, 2023-11-13:
Everything you guys are saying makes sense, including why it's complex to do per PIN.  While I understand implementation could be per pin - perhaps my uninformed ignorance is thinking about a much simpler solution.

Even though segments are volatile, if I can specify a max on a segment, then on that contiguous section of pixels, we can approximate current usage based on brightness and color.  If (brightness x color)*numPixels is over the threshold, limit the brightness of that segment that's being rendered.  

Maybe that's super naive and in the architecture there's no place to do the evaluation -- but I guess what I'm saying is even if it's not a perfect solution, having a protection at a higher level in the stack would still be helpful, especially if it will take such a large undertaking to get it correctly implemented....,

blazoncek, 2023-11-16:
Duplicate of #2364
Closing

closed 2023-11-16 by blazoncek