wled/WLED #3520: WT32-ETH01 Esp32 Sync error

yasinkryk, 2023-11-12:
### What happened?

I want to manage multiple WT32-ETH01 devices at the same time, but synchronization does not work

### To Reproduce Bug

We will have approximately 35 WT32-ETH01 devices in the project, each device will have 203 pixels 2811b LEDs.

We performed our test with 2 devices.

For your convenience, we created some presets and installed these presets on all devices from the settings menu.

When we select an animation from the Presets menu, the devices move differently from each other, sometimes they do not receive commands.

We control the LEDs segmentally, each control module has 3 segments (0-128-176-203)

and we have to complete this project within 1 week

### Expected Behavior

Synchronous operation of all devices

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.14.0 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-11-12:
> and we have to complete this project within 1 week

Hi, 
as you filed a bug report, you say there is a potential bug (software error) in WLED. So we will try to reproduce your problem (however devs usually don't have ethernet boards at hand), perform a root cause analysis, maybe implement a bugfix in the code. The fix will be scheduled for the next minor release, which _might_ come in December or January, depending on testing effort and outcomes of beta tests.

So you cannot get any bugfix within one week to support the project. It is more a matter of one or two months, or more.

If you want to get help and ideas from the WLED user community, it it best to join our [discord](https://discord.gg/QAh7wJHrRM) and ask there.

yasinkryk, 2023-11-13:
Hi, thank you for your reply, I was wondering if this was actually a malfunction in the software or if I was doing something wrong. I tried something new last night, I canceled the division into segments, I made it as a single segment and the synchronization worked, I think I will do it this way.

Additionally, if you have any information, is there any information about the maximum number of devices I can run on the same network?

closed 2024-04-21 by github-actions[bot]