wled/WLED #3524: Macro “long” option working as expected only with button 0

mechalv, 2023-11-14:
### What happened?

Hi. I have 3 push buttons, each one in a different GPIO and setup as pushbuttons, for buttons 0, 1 and 2. Then, in “Macros” i have assigned to each of the options (short, long and double) different presets, most of them cycle between different presets with the `{"ps":"1~3~"}` method.
With button 0 the “long” option works at it should; i press the button for a longer time and goes from preset 1 to 2 and so on every time i do a long press, and only jumps 1 by 1 even if i have the pushbutton pressed 1 or 5 seconds. That’s perfect, i love it!
With buttons 1 and 2 (and i have tested 3 too and changed the GPIO order just in case, no luck), if i setup the same preset and do a long press, lets say that it starts jumping infinitely and, starting from preset 1 you might fall in 2 if you are lucky, because it is very sensitive to the amount of time that you have pressed the button. If you leave it pressed and the “ps” has 20 presets (as I have in some cases) it will cycle through all of them until you release the button. The only case in which the “long” option works for me is with a “fixed/locked” preset (for example solid blue), but it is quite useless for me.

### To Reproduce Bug

1. Setup 2 pushbuttons, one for button 0 and the other for any of the other 3 buttons, lets say button 1.
2. Create 3 fixed presets (for example solid red, green and blue) and another preset to cycle through the first 3 (lets call it preset 4)
3. In macros settings put preset 4 in the “long” option for button 0, and for 1 too.
4. Test button 0 doing a long press and see how it jumps from red to green to blue each time you doo a long press, one by one, not dependent on the time you have it pressed.
5. Test button 1 by doing a long press and see the problem I’m trying to explain.

### Expected Behavior

When doing a “long” press to have buttons 1, 2 or 3 behave as button 0; jump one by one the presets, not depending on the time you have the pushbutton pressed. Please please please.

### Install Method

Self-Compiled

### What version of WLED?

0.14.0 QuinLED

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I have tested different GPIOs, button combinations, JSON and HTTP api cycle preset and i think that everything that is in my hand, just in case it is a hardware error. I have not touched in settings anything that i do not understand, is pretty much stock.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-15:
That behaviour is intentional, not a bug.

mechalv, 2023-11-15:
> That behaviour is intentional, not a bug.

Sorry, the label is added automatically, I really don't know if they are bugs or not when creating the issue, they are to me at least.
I don't understand how my problem would be a feature for somebody else, I'm no expert, but please please please, can you consider adding a checkbox to have buttons 1, 2 and/or 3 behave as button 0 when the "long" option?? One of the main purposes for using WLED in my case is the macros and the buttons. Thank you.

blazoncek, 2023-11-15:
You can custom compile and change  `WLED_LONG_PRESS` constant in `button.cpp` if needed.
It is currently set at 600ms, so a press lasting less than a second will not trigger a repeatable action.

mechalv, 2023-11-26:
> You can custom compile and change `WLED_LONG_PRESS` constant in `button.cpp` if needed. It is currently set at 600ms, so a press lasting less than a second will not trigger a repeatable action.

I have tried what you have suggested, but increasing the 600ms value does not produce what I need, it only increases the time you need to have pressed the button for WLED to detect it as a long button. But after being detected as a long press starts the repeatable actions in buttons other than 0.

I have been trying to disable the repeatable actions by changing things in the code section bellow, with any luck. I'm no programmer, the time that I have needed to get it to compile is absurd, so I just cannot imagine what to change inside "button.cpp". :disappointed:

```csharp
      if (now - buttonPressedTime[b] > WLED_LONG_PRESS) { //long press
        if (!buttonLongPressed[b]) longPressAction(b);
        else if (b) { //repeatable action (~3 times per s) on button > 0
          longPressAction(b);
          buttonPressedTime[b] = now - WLED_LONG_REPEATED_ACTION; //333ms
        }
        buttonLongPressed[b] = true;
      }
```

Leaving it as the next also makes repeatable the button 0 :sob:
```csharp
      if (now - buttonPressedTime[b] > WLED_LONG_PRESS) { //long press
        longPressAction(b);
        }
        buttonLongPressed[b] = true;
      }
```

Maybe changing something here makes the trick?? Maybe in the "case 1" line???
```csharp
  if (!macroLongPress[b]) {
    switch (b) {
      case 0: setRandomColor(col); colorUpdated(CALL_MODE_BUTTON); break;
      case 1: bri += 8; stateUpdated(CALL_MODE_BUTTON); buttonPressedTime[b] = millis(); break; // repeatable action
    }
  } else {
    unloadPlaylist(); // applying a preset unloads the playlist
    applyPreset(macroLongPress[b], CALL_MODE_BUTTON_PRESET);
  }
```

Please, I need help :pray:

mechalv, 2023-12-02:
For anybody interested or in my situation, I have found out that commenting/disabling the next lines inside "button.cpp" does the trick:
```csharp
        else if (b) { //repeatable action (~3 times per s) on button > 0
          longPressAction(b);
          buttonPressedTime[b] = now - WLED_LONG_REPEATED_ACTION; //333ms
        }
```
Now each button does only 1 "jump" while doing a long press.
My next iteration would have been writing some lines in the actions after a button is released zone, but the thing above does the trick and launches the macros while pressed, which is nicer in my opinion.

closed 2023-12-22 by mechalv