wled/WLED #3526: After saving segment with a custom name to a preset a preset cannot be saved with the default segment name to show time using scrolling text.

eoncire, 2023-11-15:
### What happened?

I have a 16x16 2d matrix set up which I am using to display text.  I created a preset with a named segment "Hello" to display that text when the preset is loaded.  I created another preset after I blanked / reset the segment name hoping to have a preset to return to the time display.  It appears that when the preset is saved with a blank segment name there is no "n" value in the preset JSON.  I have tried to modify the preset JSON before saving to add a "blank" "n" value but it did not work.

### To Reproduce Bug

Using a 2d matrix and the "Scrolling Text" effect;

1. Give the primary (only) segment a name of "Hello" and click the check box to apply.
   a. The matrix displays "Hello"
2. Create a new preset in #1 and save.
3. Erase / blank the segment name and click the check box to apply.
   b. The matrix displays the time / date
4. Create a new preset in #2 and save.
5. Load preset #1 and the matrix shows the text "Hello"
6. Load preset #2 and nothing changes.

### Expected Behavior

When loading preset #2 in the above example the time / date should be shown.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-15:
Please post preset content.

eoncire, 2023-11-15:
> Please post preset content.

This is the preset content with the "blank" segment name which should show the time.  If i load a preset with a segment named "Hello" it will display "Hello" on the matrix.  If I select the below preset after nothing will change.

{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":16,"startY":0,"stopY":16,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[8,255,0],[0,0,255],[8,255,0]],"fx":122,"sx":255,"ix":128,"pal":2,"c1":0,"c2":162,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

blazoncek, 2023-11-15:
Add `"n":""` into `"seg":[{...}]` section.
Name may not be added to conserve RAM during processing.

eoncire, 2023-11-15:
> Add `"n":""` into `"seg":[{...}]` section. Name may not be added to conserve RAM during processing.

I added that to the JSON then clicked save, but it doesn't save after the page refreshes.

What i put in.
[image]

What is there after clicking save.
[image]

eoncire, 2023-11-15:
> Add `"n":""` into `"seg":[{...}]` section. Name may not be added to conserve RAM during processing.

If I put some other text in the double quotes it will save that and display the text, but it doesn't seem to like having just the double quotes / blank.

eoncire, 2023-11-15:
Didn't mean to close this issue, clicked the wrong button...

blazoncek, 2023-11-15:
Note: when saving preset, if "Save bounds" is unchecked, segment names are not saved.

eoncire, 2023-11-15:
> Note: when saving preset, if "Save bounds" is unchecked, segment names are not saved.

Noted.  I had "Save Bounds" checked by default when saving these presets.  It looks like if the segment name is set to the default it also does NOT save the name.

blazoncek, 2023-11-15:
:grin:
Note was for myself. But I think I fixed the issue.

closed 2024-01-04 by blazoncek