wled/WLED #3528: WLED ESP8266 releases after 0.14.0-b1 won't sync other nodes when it is also acting as the AP for those nodes

clayclai, 2023-11-15:
### What happened?

I use WLED with bikes & ebikes. So, a typical installation might be one WLED node with a button on the frame running in AP mode, and two WLED nodes on the wheels running as STA attached to that AP. The group is synced from that node on the frame. This has worked fine since WLED_0.12.0_ESP8266.

The issue case is a group of 4 nodes - FRM, FL2, FW, RW, all running WLED_0.14.0-b1_ESP8266, with FRM acting as AP and syncing playlists for the group with no problem. Upgraded all nodes to WLED_0.14.0_ESP8266 via OTA, and then other nodes won't sync with FRM. But all nodes will sync from either RW, FW, or FL2. Downgrading FRM to WLED_0.14.0-b1_ESP8266 restores its ability to command the sync even though the other nodes are still at 0.14.0. Upgrading FRM to 0.14.0-b6, or 0.14.0-b4 also causes FRM to lose its ability to command the sync, but even after that, restoring it to 0.14.0-b1 also restores its ability to command the sync. But also, creating an ESP32 based FRM, running WLED_0.14.0_ESP32.bin, showed the ESP32 version doesn't share that issue. It could act as AP, and sync the other nodes even though they will still running WLED_0.14.0_ESP8266.

My preliminary conclusion is that this bug resulted in changes between 0.14.0-b1 and 0.14.0-b4 and affects only ESP8266 builds. 

### To Reproduce Bug

Create a group of ESP8266 WLED nodes running 0.14.0. Make one node the AP and have the other connect to it so that they show up in its instance list. Try to sync the group from that node. I expect it to fail. Downgrade that AP node to 0.14.0-b1. Try to sync the group from that node again. I expect it to succeed. Replace that node with an ESP32 device running 0.14.0. I expect the ability to sync the group to succeed,

### Expected Behavior

The group of nodes should continue to be synced from the node acting as AP for the group.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0_ESP8266 WLED_0.14.0-b1_ESP8266 

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-16:
Please use ESP32 for the master.

clayclai, 2023-11-16:
Thanks, that's what I figured out. Just wanted to report my observations
back to the project.

Clay Claiborne, Director
Linux Beach Bike Lights <https://linuxbeach.net/bikelights>
Venice, CA 90291
(818)964-0860

Read my blogs at the Linux Beach <http://claysbeach.blogspot.com/>
<http://wlcentral.org/user/2965/track>

blazoncek, 2023-11-16:
You will be able to sync nodes without WiFi using ESP-NOW starting with 0.15

clayclai, 2023-11-16:
That should solve a "host" of connectivity problems!

Clay

Clay Claiborne, Director
Linux Beach Bike Lights <https://linuxbeach.net/bikelights>
Venice, CA 90291
(818)964-0860

Read my blogs at the Linux Beach <http://claysbeach.blogspot.com/>
<http://wlcentral.org/user/2965/track>

closed 2024-04-21 by github-actions[bot]