wled/WLED #3530: Allow color calibration

meatheadmike, 2023-11-17:
I have a few Athom bulbs and the colors are really off compared to the various other wled devices I have. For example when I run the colorful effect with c9 2 pallette, all of my other wled devices show red, orange, yellow blue and green. But the Athom bulbs show red, yellowy-orange, geenish-yellow, blueish-green and green. I tried the Athom-specific firmware from their repo and it didn't help. I tried gamma correction and it didn't help either. I can get individual colors to show nicely by manually selecting from the color wheel. But that doesn't help in a scenario like this where there's a precanned 5-color selection. A custom pallette doesn't work here either (it reverts to 4 colors).

So the ask here is for a color calibration feature. Something like a slider that lets you apply a percentage to add or remove from each of the the RGB / WW / CW values.

Would that be feasible? Otherwise I'd also be happy if someone could point me to the place in the code where I could manually adjust the RGB values prior to output. I don't mind custom compiling.

blazoncek, 2023-11-17:
bus_manager.cpp

softhack007, 2023-11-17:
> So the ask here is for a color calibration feature.

Did you already try out "white balance correction"?

1. enable white manage in LED settings
[image]

2. move the "white balance" slider left/right until your bulb creates a real white. This slider affects all colors and palettes.
[image]

meatheadmike, 2023-11-17:
> > So the ask here is for a color calibration feature.
> 
> Did you already try out "white balance correction"?
> 
> 1. enable white management in LED settings
>    [image]
> 2. move the "white balance" slider left/right until your bulb creates a real white. This slider affects all colors and palettes.
>    [image]

Interestingly that setting seems to have zero effect on the color channels when I set the bulb up as pwm rgb+cct. When I set it up as purely pwm rgb, then it does appear to change the colors a bit. It seems to alter things about as much as the gamma correction it seems. But unfortunately thats not enough to show the orange and green correctly. And also I lose the WW and CW leds in that scenario.

Globolo001, 2024-03-06:
I got the exact same problem. Especially because the Athom Bulb has another White Channel which my WS2812's don't. So at some hues (especially blue) it's quite a bit different.
(I use the WLED sync for syncing the colours)

So it would be great if it there were some calibration option apart from WB.

dsaul560, 2024-10-23:
Same problem, different strips with wled each, have to select different hue so they look the same colors. I have hyperion and there is a color calibration option that would fix this issue.