wled/WLED #3531: Effect palette not being re-applied

vigilcoma, 2023-11-17:
### What happened?

In ver 14 when you select fire2012 fx for the first time it applies fire palette with id 35. And you can see that this palette is selected on palettes tab. After that you switch to another fx with custom palette (e.g. Fireworks Starburst). You can see that palette in UI is being updated to Rainbow. And when you back to fire2012 fx it doesn't apply Fire palette. It works with previous auto selected palette and fx looks like broken. In ver 13 everything works as expected.

### To Reproduce Bug

Select Fire2012 fx. Check palette tab, observe selected Fire palette.
Select Fireworks Starburst. Check palette tab, observe selected Rainbow palette.
Select Fire2012 fx again. Observe Rainbow palette is being selected and fx looks like broken (fire animation with ranbow colors).

### Expected Behavior

Specific palette for specific fx should be re-applied everytime the fx is selected.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-17:
"Fire 2012" effect does not include palette metadata, hence when you select the effect in UI it *does not* change existing palette. When used with "Default" palette it will internally use "Fire" palette instead.
When you select "Fireworks Starburst" using UI it has metadata that says to use "Rainbow" palette so it will switch to it (in 0_15 you have the ability to not use/disable this feature). When you switch back to "Fire 2012" no palette change happens as described above.

When used in preset, no such default behaviour is performed. It will also not load defaults if effect is selected using API call.

This behaviour is intended and not a bug. It may not be well documented but the basic info is available in [KB](https://kno.wled.ge).

I do agree that it would make sense to switch palettes on *every* UI selection but unfortunately not all effects benefit from pre-determined palettes.

vigilcoma, 2023-11-17:
> "Fire 2012" effect does not include palette metadata, hence when you select the effect in UI it _does not_ change existing palette. When used with "Default" palette it will internally use "Fire" [...]

So when you choose default palette wled select some predefined one based on current fx -  [code here](https://github.com/Aircoookie/WLED/blob/3847bfc41aa66af8b7268f685e53429109d8baae/wled00/FX_fcn.cpp#L220)
Is it correct?

And [here](https://github.com/Aircoookie/WLED/blob/3847bfc41aa66af8b7268f685e53429109d8baae/wled00/FX.cpp#L2015) we see that no palette is being applied, right?

In any case when you selet fire fx you expect some fire animation. But may see some rainbow, it looks weird.

blazoncek, 2023-11-17:
yes to both. for second it is not "applied" but "defaults". Fire 2012's legacy is seen in 1st.
weird may be personal preference. for others it may be artistic thing.
It is PITA to remove "Default" palette but eventually we may get there since *defaults* will be specified in metadata. Unless @Aircoookie decides otherwise.

vigilcoma, 2023-11-17:
Thnx.
Btw, uncheking "Use effect default parameters:" in User interface tab fixed the "issue"

blazoncek, 2023-11-17:
Yeah, I forgot it is already in. 😁

closed 2023-11-17 by vigilcoma