wled/WLED #3532: JSON API info/leds/seglc returns reserved

paulwrath1223, 2023-11-17:
### What happened?

When querying the JSON API for the list of light capabilities by segment, it returns 1 (defined as "reserved"). However, querying cfg does give the correct value for each bus. The problem seems to lie in Segment::refreshLightCapabilities, or the fact that it is only called in these 2 contexts: ```fixInvalidSegments``` and ```deserializeSegment```.

```fixInvalidSegments``` is only called when ```doInitBusses``` is true, which can only be set when buses are changes via config deserialization, or by changing settings in the settings LED tab.

I tried changing settings in the GUI and sending new configs and segments via JSON API and none updated the segment capability.

### To Reproduce Bug

1. New 14.0 installation on ESP32 (others untested)
2. Using Client mode (connected to home network via [install.wled](https://install.wled.me/)
3. change led pin in GUI settings
4. query the JSON API for info, and the list of segment capabilities says the one segment has capability "reserved"
5. adding another segment also has capability "reserved"

### Expected Behavior

The segment should have the same capability as the bus it's on. (22 or TYPE_WS2812_RGB in my case)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
http://192.168.1.40/json/info

{"ver":"0.14.0","vid":2310130,"leds":{"count":50,"pwr":748,"fps":4,"maxpwr":850,"maxseg":32,"seglc":[1,1],"lc":1,"rgbw":false,"wv":0,"cct":0},"str":false,"name":"WLED","udpport":21324,"live":false,"liveseg":-1,"lm":"","lip":"","ws":0,"fxcount":187,"palcount":71,"cpalcount":0,"maps":[{"id":0}],"wifi":{"bssid":"FC:EC:DA:A4:C4:77","rssi":-61,"signal":78,"channel":1},"fs":{"u":12,"t":983,"pmt":1700257440},"ndc":6,"arch":"esp32","core":"v3.3.6-16-gcc5440f6a2","lwip":0,"freeheap":197860,"uptime":285,"time":"2023-11-17, 21:46:41","opt":79,"brand":"WLED","product":"FOSS","mac":"a842e38d9828","ip":"192.168.1.40"}
```

### Anything else?

I kind of need this functionality for a project, and I'd be happy to help fix it

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paulwrath1223, 2023-11-18:
Ok, so after digging through the source, the capability returned by the API call is actually a different structure than the bus light capabilities. Not only that, but it actually returns the sum of all capabilities shared by all buses, so even if the segment is on a single channel strip, if the WLED has any RGB buses, it will say the strip has RGB support. I still think the desired behavior should be returning the capability of the bus it's on, but oh well.

closed 2023-11-18 by paulwrath1223

blazoncek, 2023-11-18:
Segment has its own capabilities. It is available in the state object.

paulwrath1223, 2023-11-18:
I'm sorry if this is super obvious and I'm just missing it, but I'm looking at the state and segment serialization functions, as well as the Segment class itself. But I just can't find it. Which field is it?

blazoncek, 2023-11-18:
I beg your pardon, I mixed things up. It is the `seglc` array in `info` object, not `state`.