wled/WLED #3534: flashing from WLED_0.14.0_ESP01.bin on ESP-01S faulty.

Jens869, 2023-11-18:
### What happened?

I flashed to ESP-01S with "ESP flasher" WLED_0.14.0_ESP01.bin. After that I used https://install.wled.me/ to enter the Wifi settings.  As a result, the web interface cannot be reached even though the router has assigned a fixed IP.
Even searching for another IP didn't bring a solution.

Then I have WLED_0.14.0-b6_ESP01.bin in the same way flashed. Then the web interface is not accessible.

Then I flashed WLED_0.14.0_ESP02.bin in the same way. Then the web interface is accessible but the Flash is too small.

Then I flashed WLED_0.13.3_ESP01.bin in the same way. Then the web interface is accessible and the flash size is correct.

WLED_0.14.0-b6_ESP01.bin andWLED_0.14.0_ESP01.bin are incorrect.

OS: Mac OS 11.7.10 Intel 

### To Reproduce Bug

use WLED_0.13.3_ESP01.bin until bug is fixed

### Expected Behavior

fix [WLED_0.14.0_ESP01.bin](https://github.com/Aircoookie/WLED/releases/download/v0.14.0/WLED_0.14.0_ESP01.bin)
fix web interface for ESP-021 S 1M

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0_ESP01.bin

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-18:
Unfortunately you cannot configure ESP01 using web installer.
It is only possible to flash firmware.
The other option is to use some offline flasher like esptool.

Jens869, 2023-11-19:
Yes, that's why I use ESP Flasher (based on NodeMCU Pyflasher)
In the second step, the WiFi can then be set using the web installer.

I have now flashed WLED_0.14.0_ESP01.bin again onto the ESP-01S with ESP Flasher.
Then I switched the Wifi to WLED acesspoint.
There I initially assigned a self-assigned IP address.
After 1/4 hour I got the IP and was able to open 4.3.2.1

There I could see the web interface.
I assigned a fixed IP and entered the WiFi data.

Then I switched to my home WiFi and called up the IP address. I also started a ping.
Result: the WLED website cannot be accessed.
The ping is very slow to reach at first and then no longer accessible. Request timeout for icmp_seq 

Then I did it all over again with WLED_0.14.0_ESP02.bin. 
So the firmware for the 2 MB chip on the 1MB chip. That works but OTA will break it.

Jens869, 2023-11-19:
> Unfortunately you cannot configure ESP01 using web installer. It is only possible to flash firmware. The other option is to use some offline flasher like esptool.

No. you can not flash an ESP01 using web installer.  It will always write down the 4MB firmware.
But configure Wifi on an ESP01 using web installer in step 2 work fine

softhack007, 2023-11-19:
The wled 0.14 binary is too big for updating esp01 (0.5 or 1M flash) over OTA. Also I don't see how the web installer could be "fixed" to work the way you want it.

The wled binary need about 850kb now, plus at least 128kb  for filesystem. So 1MB flash is "on the edge", and it could be soon too small, if the next release grows just a few kbytes more.

Closing the report, as I don't see a way to support 1MB chips in future. Please feel free to re-open, in case you have a clever idea an your willing to implement it (PR).

closed 2023-11-19 by softhack007

softhack007, 2023-11-19:
> I do not think so. The WLED_0.14.0_ESP01.bin is 834 KB big and brocken. The WLED_0.14.0_ESP02.bin is 847 KB in size and works. It can't be the size.

Below the build environments for ESP01 and ESP02. As you can see, the main differences are `board`, `ldscripts` and `WLED_DISABLE_OTA`. Other than that, source code for esp01 and esp02 is exactly the same. Please tell me how to "fix" that.

https://github.com/Aircoookie/WLED/blob/1dab26bcbcac051f2b7be47a2d5c757a9938bf1f/platformio.ini#L339-L357